Samsung/TizenRT #2458: I would like to know about the use of the `st_things_deinitialize` function.

JunhwanPark, 2018-11-19:
In the `st_things` examples, I can see that the `st_things_initialize` function is being used. However, I could not find **deinitialize** which is expected to be used after **initialize**.

for example: sample_blink_things.c )
```c
int ess_process(void)
{
	bool easysetup_complete = false;
	st_things_initialize("device_def.json", &easysetup_complete);

	blink_init();
	blink_start();
	st_things_register_request_cb(handle_get_request, handle_set_request);
	st_things_register_user_confirm_cb(handle_ownership_transfer_request);
	st_things_register_things_status_change_cb(handle_things_status_change);
	if (st_things_start() == ST_THINGS_ERROR_NONE) {
		printf("[%s]=====================================================\n", TAG);
		printf("[%s]                    Stack Started                    \n", TAG);
		printf("[%s]=====================================================\n", TAG);
	}

	return 0;
}
```
I would like to know about the use of the `st_things_deinitialize` function.

SenthilKumarGS, 2018-11-26:
@JunhwanPark 
st_things_deinitialize() is supposed to be called during app termination which is not handled by the samples..

JunhwanPark, 2018-11-26:
@SenthilKumarGS 
The samples included in TizenRT do not call `st_things_deinitialize()` functions. If possible, include it in the sample to use the `st_things_deinitialize()` function.

SenthilKumarGS, 2018-11-27:
Sure @JunhwanPark

SenthilKumarGS, 2018-12-07:
@Hyunjun85 & @JeongJunSik, 

I need your opinion. It's very important.

**Background:-**
1 year ago, public API (st_things_stop) was removed because it was not needed. 
PR: https://github.com/Samsung/TizenRT/pull/683#discussion_r141527637

**Issue:-**
st_things_deinitialize() is not doing complete cleanup as it doesn't destroy iotivity resources and threads. So if we provide a usage scenario for this function in the application, then there will be a problem if developers think that the resource usage of st_things will be freed after calling st_things_deinitialize() which is not happening currently. 

**Solution:-**
So we need to bring back the public API (st_things_stop). Then we can clearly show how st_things_stop() and st_things_deinitialize() can be used. 
(Ex: By adding command-line argument in the sample to stop. 
st_things_sample -> Starts ST_Things application.
st_things_sample stop -> Stops ST_Things application.)

Hyunjun85, 2019-04-03:
When we have discussion about this. we can not find out any requirements. 
And we have the pr(it was closed). but it need to get some communicate with stakeholders more. 
https://github.com/Samsung/TizenRT/pull/2573

closed 2019-04-03 by Hyunjun85