Samsung/TizenRT #2459: Object files remain after `make clean`.

JunhwanPark, 2018-11-19:
I have tested it like this and want to delete all object files after build using the `make clean` command.
```bash
os/tools$ sh configure.sh artik053/st_things
os$ make
 <...>
os$ make clean
 <...>
```

However, the following files remained.
```bash
os$ find .. -name "*.o"
../apps/examples/st_things/light/sample_light_user.o
../apps/examples/st_things/light/sample_light_things.o
../os/arch/arm/src/artik053/src/artik053_pwm.o
../os/arch/arm/src/artik053/src/artik053_progmem.o
../os/arch/arm/src/artik053/src/artik053_boot.o
../os/arch/arm/src/artik053/src/artik053_wlan.o
../os/arch/arm/src/artik053/src/artik053_tash.o
../os/arch/arm/src/artik053/src/boardctl.o
```

SenthilKumarGS, 2018-11-22:
@harish-191 
Check this issue.

harish-191, 2018-11-22:
I fixed this issue and created the PR. It is in review.

SenthilKumarGS, 2018-11-22:
@Hyunjun85 @sangwon03 
To remove the object files under **os** directory,  could you please inform the person in-charge?

JunhwanPark, 2018-12-24:
in b73e522da9d96e62dedda59329e4ac7c56a07025
```bash
$ find .. -name "*.o"
../apps/examples/st_things/light/sample_light_user.o
../apps/examples/st_things/light/sample_light_things.o
```

JunhwanPark, 2019-01-16:
@harish-191 
I have confirmed that all object files are removed. Next, let us know what PR has solved this issue.

closed 2019-01-16 by JunhwanPark