Samsung/TizenRT #2460: `libcoap.a` file should be removed with `make clean` command

JunhwanPark, 2018-11-19:
I tested it as follows.

```bash
$ git clone https://github.com/Samsung/TizenRT
$ cd TizenRT
TizenRT$ find . -name "*.a"
./external/WICED/libbcmexternal.a
./os/board/cy4390x/src/libboard_noaudio.a
./os/board/cy4390x/src/libboard_audio.a
./os/arch/arm/src/s5j/sss/libispdriver.a
./os/arch/arm/src/bcm4390x/libbcmarch.a
```
After the build, I wanted to delete the object files and library files I created using the `make clean` command.
```bash
os/tools$ sh configure.sh artik053s/st_things
os$ make
 <...>
os$ make clean
 <...>
os$ find .. -name "*.a"
../external/WICED/libbcmexternal.a
../external/iotivity/iotivity_1.2-rel/extlibs/libcoap/libcoap.a
../os/board/cy4390x/src/libboard_noaudio.a
../os/board/cy4390x/src/libboard_audio.a
../os/arch/arm/src/s5j/sss/libispdriver.a
../os/arch/arm/src/bcm4390x/libbcmarch.a
```

I found this file (`../external/iotivity/iotivity_1.2-rel/extlibs/libcoap/libcoap.a`) to be left even though I cleaned the library files. There is no problem in operation even if this file is deleted.

SenthilKumarGS, 2018-11-19:
PR: https://github.com/Samsung/TizenRT/pull/2463

SenthilKumarGS, 2018-11-19:
@Hyunjun85 @juitem ,
Kindly review and approve #2463.

JunhwanPark, 2018-11-22:
@SenthilKumarGS  Thank you for making patch.

closed 2018-11-22 by JunhwanPark