Samsung/TizenRT #2481: Artik 055s fall into infinite loop when calling pow() with specific value

dellhart, 2018-11-23:
Related post: [thread in artitk forum](https://developer.artik.io/forums/t/artik-055s-halt-when-calling-pow-with-specific-value/7598)

I'm not sure here is right place to post SDK related(ARTIK 055s, SDK version is 1.7) issue, so if I am wrong, please let me know.

As title indicates, calling the function **double pow(double x)** with specific x make Artik 055s fall into infinite loop. One of the value that re-produce this bug is {0x00, 0x00, 0x00, 0x00, 0x80, 0xF8, 0xED, 0xBF} which is approximately -0.936584, therefore, if you want to reproduce that bug in your board, you need to use malloc for prevent lose precision.

```c
double d = 0;
double result;
unsigned char dbuf[sizeof(d)] = {0x00, 0x00, 0x00, 0x00, 0x80, 0xF8, 0xED, 0xBF};
memcpy(&d, dbuf, sizeof(d));

result = pow(d) // can't escape from this line
```

Also, Here is code snippet that use for tracking bug.
[code](https://gist.github.com/dellhart/8a6fd30e824405cc81c1cad710247544)
In this code, mlog and mpow had copied from lib/libc/math/lib_*.c in this repository, means, Samsung/TizenRT.

After I trace execution order that pow() make, I found that pow(d) hangs because calling log(-d), for calculate pow(d). And i have no idea that what exactly line of codes are make artik hangs yet.

sunghan-chang, 2018-11-26:
@dellhart Thank you for raising issue. We are under investigation and let you know a result.

sunghan-chang, 2018-12-07:
@dellhart This is fixed by #2496 . Could you check yours with that?

dellhart, 2018-12-07:
@sunghan-chang Please let me notify the result after i change the build environment to this codebase. due to I'm using Artik IDE + SDK combo now so can't apply this change to my end.

sunghan-chang, 2019-04-03:
@dellhart It's old issue so that let's close now. And when you have an issue, could you reopen this or open new one?

closed 2019-04-03 by sunghan-chang