Samsung/TizenRT #2527: Mater branch build error - git log : commit f6d6e69339c2b2eb9a4d86ac98026c64eadfe1c2

juhaki, 2018-12-06:
juhaki@juhakidesk3:~/workspace/181030TizenRT_SR_onytest1$ git log
commit f6d6e69339c2b2eb9a4d86ac98026c64eadfe1c2
Merge: 75e4a66 1c16e23
Author: Hyunjun Kim <36025642+Hyunjun85@users.noreply.github.com>
Date:   Thu Dec 6 14:55:37 2018 +0900

    Merge pull request #2525 from SenthilKumarGS/remove_stop_softap
    
    framework/st_things: Remove unused "Stop SoftAP from Cloud" feature.

commit 75e4a66db2658793b8cbcae4dbf7c9181fb66d8d
Merge: 803d87f c68431f
Author: Hyunjun Kim <36025642+Hyunjun85@users.noreply.github.com>
Date:   Thu Dec 6 14:51:56 2018 +0900

    Merge pull request #2467 from harish-191/S18TRST-23
    
    framework/st_things : Add EncType and SecType values for AP scan list
...
juhaki@juhakidesk3:~/workspace/181030TizenRT_SR_onytest1$ cat .git/config 
[core]
	repositoryformatversion = 0
	filemode = true
	logallrefupdates = true
[remote "origin"]
	url = https://github.com/Samsung/TizenRT
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master
juhaki@juhakidesk3:~/workspace/181030TizenRT_SR_onytest1$ git branch
* master
juhaki@juhakidesk3:~/workspace/181030TizenRT_SR_onytest1$ 

...
arm-none-eabi-gcc -c -fno-builtin -mcpu=cortex-r4 -mfpu=vfpv3 -Wall -Werror -Wstrict-prototypes -Wshadow -Wundef -Wno-implicit-function-declaration -Wno-unused-function -Wno-unused-but-set-variable -g -Os -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer  -I. -isystem /home/juhaki/workspace/181030TizenRT_SR_onytest1/os/include -isystem /home/juhaki/workspace/181030TizenRT_SR_onytest1/os/../framework/include -isystem /home/juhaki/workspace/181030TizenRT_SR_onytest1/os/../external/include   -pipe -ffunction-sections -fdata-sections  mm_heap/mm_zalloc.c -o  bin/mm_zalloc.o
CC:  mm_heap/mm_heap_regioninfo.c
arm-none-eabi-gcc -c -fno-builtin -mcpu=cortex-r4 -mfpu=vfpv3 -Wall -Werror -Wstrict-prototypes -Wshadow -Wundef -Wno-implicit-function-declaration -Wno-unused-function -Wno-unused-but-set-variable -g -Os -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer  -I. -isystem /home/juhaki/workspace/181030TizenRT_SR_onytest1/os/include -isystem /home/juhaki/workspace/181030TizenRT_SR_onytest1/os/../framework/include -isystem /home/juhaki/workspace/181030TizenRT_SR_onytest1/os/../external/include   -pipe -ffunction-sections -fdata-sections  mm_heap/mm_heap_regioninfo.c -o  bin/mm_heap_regioninfo.o
mm_heap/mm_heap_regioninfo.c:5:51: error: expected expression before '}' token
 void *regionx_start[CONFIG_MM_REGIONS] = {(void *)};
                                                   ^
Makefile:102: recipe for target 'bin/mm_heap_regioninfo.o' failed
make[1]: *** [bin/mm_heap_regioninfo.o] Error 1
make[1]: Leaving directory '/home/juhaki/workspace/181030TizenRT_SR_onytest1/os/mm'
LibTargets.mk:249: recipe for target 'mm/libmm.a' failed
make: *** [mm/libmm.a] Error 2
[STATUS] FAILED

jeongchanKim, 2018-12-07:
Which config did you try to build? I tested with latest code for artik053/tc, artik053/st_things, but there is no build error.
Please share your configuration and test guide.

jeongchanKim, 2018-12-07:
I checked your .config file. I think you had a mistake for applying changes.
In your config file, there are
```
CONFIG_RAM_START=0x02023800
CONFIG_RAM_SIZE=968704
#CONFIG_RAM_REGIONx_START="0x00000000,"
#CONFIG_RAM_REGIONx_SIZE="0,"
```
CONFIG_RAM_START and CONFIG_RAM_SIZE are not used anymore. 
They are changed to CONFIG_RAM_REGIONx_START and SIZE.

You can set like below:
CONFIG_RAM_REGIONx_START="0x02023800"
CONFIG_RAM_REGIONx_SIZE="968704"

juhaki, 2018-12-07:
curretly.. latest commit id is "87ac1a57d93819497bdd26b5ff71d1d8d10e5d6e".
Also. the commit is build succeed.

=====================================================
commit 87ac1a57d93819497bdd26b5ff71d1d8d10e5d6e
Merge: fff4071 b502938
Author: sunghan-chang <sh924.chang@samsung.com>
Date:   Fri Dec 7 14:27:51 2018 +0900

    Merge pull request #2496 from drashti304/mathlib
    
    libc/math: Update log() and logf() functions
...
==========================================

closed 2018-12-07 by juhaki