Samsung/TizenRT #2860: NXP imxrt1050evkb binary download and boot up

kishore-sn, 2019-03-29:
I am trying to download TizenRT binary built with imxrt1050-evk/nxp_demo config onto the NXP 1050 evkb board. But when we boot the board our changes are not reflected. It seems that the binary download is not successful. So, please provide below information.

- What are the exact steps to download binary?
- How does the drag and drop function work?
- What is the boot up architecture for imxrt project?
- Is the boot up done from hyper flash or spi flash? What are the exact boot up steps before the kernel gets started?

Also, please help me for this tizenRT protected build issue:
- We need to download 2 binaries for protected build. User and kernel binary needs to be flashed in uflash and kflash region respectively. Can you please provide a tool to perform this?

sunghan-chang, 2019-03-29:
@HarrisonChoi750405 Could you check this?

HarrisonChoi750405, 2019-03-29:
What are the exact steps to download binary?
: I am currently using the following two ways
 1) select the d menu in build menu when using dbuild menu
 2) copy tinyara.hex file to RT1050-EVK folder that is detected automatically when connecting iMXRT1050-evk by using USB cable

How does the drag and drop function work?
: Can you explain what is the drag and drop function?

What is the boot up architecture for imxrt project?
: Can you explain what you are wondering in detail?

Is the boot up done from hyper flash or spi flash? What are the exact boot up steps before the kernel gets started?
: In case of iMXRT1050-evk, we are currently using hyper flash for boot-up.
 Can you explain what you are wondering in detail?

At first, I have no experience about two flash binary that you are saying.
I think that I need to ask the main engineer about it after two binary can be made.
In my side, can you tell me how to make two binary?

Best Regards
Harrison

kishore-sn, 2019-03-29:
What are the exact steps to download binary?
: I tried below steps
- Add printf("Hello world\n") in nxp_demo_config.c
- Build with imxrt1050-evk/nxp_demo config
- Connect board.  RT1050-EVK drive is shown
- copy tinyara.hex to  RT1050-EVK drive
- wait for 30 seconds
**-  RT1050-EVK drive goes away. 
-  RT1050-EVK does not come back**
- Try to disconnect and reconnect board
- Now  RT1050-EVK and ttyACM0 interface is present
- Check serial logs. My changes for "Hello world" are not reflected
- This means my binary is not downloaded properly
- Please let me know how to fix this

Initially, when I copy tinyara.hex to  RT1050-EVK, it generated Fail.txt. 
The message was : "In application programming aborted due to an out of bounds address"

Then I performed below steps:
- Press and hold reset button
- Connect board to the USB debug port
- MAINTENANCE drive is opened
- Copy  "DAPLink v0244 - Supports onboard spi-flash" from "https://www.nxp.com/support/developer-resources/run-time-software/kinetis-developer-resources/ides-for-kinetis-mcus/opensda-serial-and-debug-adapter:OPENSDA?&tid=vanOpenSDA#MIMXRT1050-EVK"  to MAINTENANCE drive
- Board rebooted and showed RT1050-EVK drive

HarrisonChoi750405, 2019-03-29:
Please let me know the version that you are testing.
I want to test it with your version in my side.

I have ever met the same problem that Fail.txt is shown in RT1050-EVK folder and downloading failed.
But it is strange for me that you always meet the same problem.

In my case, I can see the normal starting log after copying tinyara.hex and waiting for dectecting RT1050-EVK foloder again and pushing reset button.

kishore-sn, 2019-03-30:
I am testing the latest code in Github master branch. 
I just tried to add a printf statement in nxp demo file. 

I am not getting the Fail.txt every time. It happened only some times. But after that I am not able to download the new binary. 

When I copy tinyara.hex to RT1050-EVK folder, after some time the RT1050-EVK folder does not come back at all.

HarrisonChoi750405, 2019-04-01:
I am meeting the same problem.
I asked NXP's main support team about the another way to download built image.
I will share the guidance if I have any answer from him.

manoharahk, 2019-04-03:
Hello @HarrisonChoi750405

Is there any flash programming tool  available for NXP IMXRT-10xx series to program binaries directly at specific flash offset?

sunghan-chang, 2019-05-02:
Here is a guidance of flashing.
In IMXRT1050RM.PDF.

8.3.1 Boot mode pin settings
The device has four boot modes (one is reserved for NXP use). The boot mode is selected
based on the binary value stored in the internal BOOT_MODE register.
The BOOT_MODE is initialized by sampling the BOOT_MODE0 and BOOT_MODE1
inputs on the rising edge of the POR_B. After these inputs are sampled, their subsequent
state does not affect the contents of the BOOT_MODE internal register. The state of the
internal BOOT_MODE register may be read from the BMOD[1:0] field of the SRC Boot
Mode Register (SRC_SBMR2). The available boot modes are: Boot From Fuses, serial
boot via USB, and Internal Boot. See this table for settings:

Table 8-2. Boot MODE pin settings

BOOT_MODE[1:0] Boot Type
00 Boot From Fuses
**01 Serial Downloader**
10 Internal Boot
11 Reserved

closed 2019-05-02 by sunghan-chang