Samsung/TizenRT #2969: os/kernel: pthread_mutex_* functions is called in non-pthread task

zhouxinhe, 2019-04-18:
I can find codes (invoking pthread_mutex_(un)lock() in non-pthread task) in TizenRT:
when pthread_mutex_lock() is called, pthread_mutex_add() would be  executed (in case CONFIG_PTHREAD_MUTEX_UNSAFE isn't defined)...
```c
static void pthread_mutex_add(FAR struct pthread_mutex_s *mutex)
{
	FAR struct pthread_tcb_s *rtcb = (FAR struct pthread_tcb_s *)this_task();
	irqstate_t flags;

	DEBUGASSERT(mutex->flink == NULL);

	/* Add the mutex to the list of mutexes held by this task */

	flags = irqsave();
	mutex->flink = rtcb->mhead;
	rtcb->mhead = mutex;
	irqrestore(flags);
}
```
in above code, it assumes that current task is a pthread task.
but it's not sure, so I try to add below assertion code in pthread_mutex_add():
```c
DEBUGASSERT((rtcb->cmn.flags & TCB_FLAG_TTYPE_MASK) == TCB_FLAG_TTYPE_PTHREAD);
```
then I found asstion as below:
```log
## Starting application at 0x040C8020 ...
s5j_sflash_init: FLASH Quad Enabled
up_assert: Assertion failed at file:pthread/pthread_mutex.c line: 93 task: Idle Task
```
It's related to file os\drivers\wireless\scsccm_if.c (wifi is enabled).
--- Idle task is not a pthread task!
when I disabled wifi and then I got same assertion in example app code.
--- "appmain" task is not a pthread task!
I can not list all assertioins here...

My questions:
**pthread_mutex_ functions should be only called in pthread task, right?**
- if yes, I suggest to add `DEBUGASSERT` above and then fix the assertions.
- otherwise, there's problem:
Any change in task management fields of struct pthread_tcb_s will damage the task management fields of real struct task_tcb_s (beause current task is non-pthread task);
or maybe overwrite memory of next neighboring node (e.g. `rtcb->mhead`'s offset may be larger than sizeof(struct task_tcb_s)).

zhouxinhe, 2019-04-18:
@sunghan-chang

zhouxinhe, 2019-04-18:
I checked latest code of Nuttx,
It's fixed as below:
```c
static void pthread_mutex_add(FAR struct pthread_mutex_s *mutex)
{
  FAR struct tcb_s *rtcb = this_task();

  DEBUGASSERT(mutex->flink == NULL);

  /* Check if this is a pthread.  The main thread may also lock and unlock
   * mutexes.  The main thread, however, does not participate in the mutex
   * consistency logic.  Presumably, when the main thread exits, all of the
   * child pthreads will also terminate.
   *
   * REVISIT:  NuttX does not support that behavior at present; child pthreads
   * will persist after the main thread exits.
   */

  if ((rtcb->flags & TCB_FLAG_TTYPE_MASK) == TCB_FLAG_TTYPE_PTHREAD)
    {
      FAR struct pthread_tcb_s *ptcb = (FAR struct pthread_tcb_s *)rtcb;
      irqstate_t flags;

      /* Add the mutex to the list of mutexes held by this pthread */

      flags        = enter_critical_section();
      mutex->flink = ptcb->mhead;
      ptcb->mhead  = mutex;
      leave_critical_section(flags);
    }
}
```

xixidodo, 2019-04-22:
if so, then it is really risky to  invoke pthread_mutex_* in non-pthread task, causing undefined behavior which is hard to find the root cause.
Incorrect usage shall be fixed, and protection shall be added.

sunghan-chang, 2019-04-26:
@zhouxinhe @xixidodo Next week, I will check all of pthread APIs and will update them.

sunghan-chang, 2019-04-26:
@zhouxinhe Thank you for notice