Samsung/TizenRT #3123: esp32_DevKitC arch/stdarg.h: No such file or directory

ChiShaoJun, 2019-05-14:
Scons installed.  
select "Menuconfig" External Libraries->enable / disable iotivity stack  
select "Build with Current Configurations" Error  
log  
`Docker Image Version : 1.5.0
echo "create .version file"; \
tools/version.sh -v 2.0 .version; \
chmod 755 .version
create .version file
LN: include/arch/board to board/esp32_devkit/include
LN: include/arch/chip to arch/xtensa/include/esp32
LN: arch/xtensa/src/chip to arch/xtensa/src/esp32
LN: include/apps to ../apps/include
Previous Build Outputs - ../build/output/bin - are deleted
make[1]: Entering directory '/root/tizenrt/apps'
make[2]: Entering directory '/root/tizenrt/apps/examples'
make[3]: Entering directory '/root/tizenrt/apps/examples/mtdpart'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/mtdpart'
make[3]: Entering directory '/root/tizenrt/apps/examples/ntpclient_test'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/ntpclient_test'
make[3]: Entering directory '/root/tizenrt/apps/examples/mqtt_test'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/mqtt_test'
make[3]: Entering directory '/root/tizenrt/apps/examples/dtls_server'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/dtls_server'
make[3]: Entering directory '/root/tizenrt/apps/examples/kernel_sample'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/kernel_sample'
make[3]: Entering directory '/root/tizenrt/apps/examples/mpu'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/mpu'
make[3]: Entering directory '/root/tizenrt/apps/examples/libcoap_client'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/libcoap_client'
make[3]: Entering directory '/root/tizenrt/apps/examples/proc_test'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/proc_test'
make[3]: Entering directory '/root/tizenrt/apps/examples/grpc_greeter_client'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/grpc_greeter_client'
make[3]: Entering directory '/root/tizenrt/apps/examples/tls_selftest'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/tls_selftest'
make[3]: Entering directory '/root/tizenrt/apps/examples/tls_benchmark'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/tls_benchmark'
make[3]: Entering directory '/root/tizenrt/apps/examples/iotjs_startup'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/iotjs_startup'
make[3]: Entering directory '/root/tizenrt/apps/examples/sensorbd_demo'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/sensorbd_demo'
make[3]: Entering directory '/root/tizenrt/apps/examples/protobuf'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/protobuf'
make[3]: Entering directory '/root/tizenrt/apps/examples/iotivity_simpleserver'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/iotivity_simpleserver'
make[3]: Entering directory '/root/tizenrt/apps/examples/cxxtest'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/cxxtest'
make[3]: Entering directory '/root/tizenrt/apps/examples/st_things'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/st_things'
make[3]: Entering directory '/root/tizenrt/apps/examples/task_manager_sample'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/task_manager_sample'
[... 524 lines cut ...]
Makefile:143: recipe for target '.depend' failed
make[1]: *** [.depend] Error 1
make[1]: Leaving directory '/root/tizenrt/os/mm'
make[1]: Entering directory '/root/tizenrt/os/fs'
make[1]: Nothing to be done for 'depend'.
make[1]: Leaving directory '/root/tizenrt/os/fs'
make[1]: Entering directory '/root/tizenrt/os/drivers'
make[1]: Nothing to be done for 'depend'.
make[1]: Leaving directory '/root/tizenrt/os/drivers'
make[1]: Entering directory '/root/tizenrt/os/net'
make[1]: Nothing to be done for 'depend'.
make[1]: Leaving directory '/root/tizenrt/os/net'
make[1]: Entering directory '/root/tizenrt/os/kernel'
CC:  init/os_start.c
In file included from /root/tizenrt/os/include/tinyara/logm.h:29:0,
                 from /root/tizenrt/os/include/debug.h:72,
                 from init/os_start.c:61:
/root/tizenrt/os/include/stdarg.h:69:25: fatal error: arch/stdarg.h: No such file or directory
compilation terminated.
Makefile:94: recipe for target 'os_start.o' failed
make[1]: *** [os_start.o] Error 1
make[1]: Leaving directory '/root/tizenrt/os/kernel'
LibTargets.mk:79: recipe for target 'kernel/libkernel.a' failed
make: *** [kernel/libkernel.a] Error 2
`

ChiShaoJun, 2019-05-14:
I added stdarg.h file in ~/TizenRT/os/arch/xtensa/include
```
/* Copyright (C) 1989-2015 Free Software Foundation, Inc.

This file is part of GCC.

GCC is free software; you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation; either version 3, or (at your option)
any later version.

GCC is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

Under Section 7 of GPL version 3, you are granted additional
permissions described in the GCC Runtime Library Exception, version
3.1, as published by the Free Software Foundation.

You should have received a copy of the GNU General Public License and
a copy of the GCC Runtime Library Exception along with this program;
see the files COPYING3 and COPYING.RUNTIME respectively.  If not, see
<http://www.gnu.org/licenses/>.  */

/*
 * ISO C Standard:  7.15  Variable arguments  <stdarg.h>
 */

#ifndef _STDARG_H
#ifndef _ANSI_STDARG_H_
#ifndef __need___va_list
#define _STDARG_H
#define _ANSI_STDARG_H_
#endif /* not __need___va_list */
#undef __need___va_list

/* Define __gnuc_va_list.  */

#ifndef __GNUC_VA_LIST
#define __GNUC_VA_LIST
typedef __builtin_va_list __gnuc_va_list;
#endif

/* Define the standard macros for the user,
   if this invocation was from the user program.  */
#ifdef _STDARG_H

#define va_start(v,l)	__builtin_va_start(v,l)
#define va_end(v)	__builtin_va_end(v)
#define va_arg(v,l)	__builtin_va_arg(v,l)
#if !defined(__STRICT_ANSI__) || __STDC_VERSION__ + 0 >= 199900L || defined(__GXX_EXPERIMENTAL_CXX0X__)
#define va_copy(d,s)	__builtin_va_copy(d,s)
#endif
#define __va_copy(d,s)	__builtin_va_copy(d,s)

/* Define va_list, if desired, from __gnuc_va_list. */
/* We deliberately do not define va_list when called from
   stdio.h, because ANSI C says that stdio.h is not supposed to define
   va_list.  stdio.h needs to have access to that data type, 
   but must not use that name.  It should use the name __gnuc_va_list,
   which is safe because it is reserved for the implementation.  */

#ifdef _BSD_VA_LIST
#undef _BSD_VA_LIST
#endif

#if defined(__svr4__) || (defined(_SCO_DS) && !defined(__VA_LIST))
/* SVR4.2 uses _VA_LIST for an internal alias for va_list,
   so we must avoid testing it and setting it here.
   SVR4 uses _VA_LIST as a flag in stdarg.h, but we should
   have no conflict with that.  */
#ifndef _VA_LIST_
#define _VA_LIST_
#ifdef __i860__
#ifndef _VA_LIST
#define _VA_LIST va_list
#endif
#endif /* __i860__ */
typedef __gnuc_va_list va_list;
#ifdef _SCO_DS
#define __VA_LIST
#endif
#endif /* _VA_LIST_ */
#else /* not __svr4__ || _SCO_DS */

/* The macro _VA_LIST_ is the same thing used by this file in Ultrix.
   But on BSD NET2 we must not test or define or undef it.
   (Note that the comments in NET 2's ansi.h
   are incorrect for _VA_LIST_--see stdio.h!)  */
#if !defined (_VA_LIST_) || defined (__BSD_NET2__) || defined (____386BSD____) || defined (__bsdi__) || defined (__sequent__) || defined (__FreeBSD__) || defined(WINNT)
/* The macro _VA_LIST_DEFINED is used in Windows NT 3.5  */
#ifndef _VA_LIST_DEFINED
/* The macro _VA_LIST is used in SCO Unix 3.2.  */
#ifndef _VA_LIST
/* The macro _VA_LIST_T_H is used in the Bull dpx2  */
#ifndef _VA_LIST_T_H
/* The macro __va_list__ is used by BeOS.  */
#ifndef __va_list__
typedef __gnuc_va_list va_list;
#endif /* not __va_list__ */
#endif /* not _VA_LIST_T_H */
#endif /* not _VA_LIST */
#endif /* not _VA_LIST_DEFINED */
#if !(defined (__BSD_NET2__) || defined (____386BSD____) || defined (__bsdi__) || defined (__sequent__) || defined (__FreeBSD__))
#define _VA_LIST_
#endif
#ifndef _VA_LIST
#define _VA_LIST
#endif
#ifndef _VA_LIST_DEFINED
#define _VA_LIST_DEFINED
#endif
#ifndef _VA_LIST_T_H
#define _VA_LIST_T_H
#endif
#ifndef __va_list__
#define __va_list__
#endif

#endif /* not _VA_LIST_, except on certain systems */

#endif /* not __svr4__ */

#endif /* _STDARG_H */

#endif /* not _ANSI_STDARG_H_ */
#endif /* not _STDARG_H */

```
also error

xixidodo, 2019-05-14:
@ChiShaoJun  Glad that you are interested in TizenRT.
IoTivity/SmartThings on ESP32 is not released yet, and we are working on it now on esp_wrover_kit.
It is estimated that it shall be available soon, maybe end of this month.

We will keep you posted with related PR here.

zhouxinhe, 2019-05-14:
@ChiShaoJun 
Hello, I also encountered this building error, I solved it as you did, and it's working fine.
Please make clean and try again.

ChiShaoJun, 2019-05-15:
1.git pull
2.I added stdarg.h file in ~/TizenRT/os/arch/xtensa/include
3.Build Dist-Clean
4.Menuconfig
5.Build with Current Configurations
error log
```
Docker Image Version : 1.5.0
make[1]: Entering directory '/root/tizenrt/os/tools'
make[1]: Leaving directory '/root/tizenrt/os/tools'
echo "create .version file"; \
tools/version.sh -v 2.0 .version; \
chmod 755 .version
create .version file
make[1]: Entering directory '/root/tizenrt/os/tools'
make[1]: Leaving directory '/root/tizenrt/os/tools'
LN: include/arch to arch/xtensa/include
LN: include/arch/board to board/esp32_devkit/include
LN: include/arch/chip to arch/xtensa/include/esp32
LN: arch/xtensa/src/board to board/esp32_devkit/src
LN: arch/xtensa/src/chip to arch/xtensa/src/esp32
LN: include/apps to ../apps/include
make[1]: Entering directory '/root/tizenrt/apps'
make[2]: Entering directory '/root/tizenrt/apps/examples'
make[3]: Entering directory '/root/tizenrt/apps/examples/mtdpart'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/mtdpart'
make[3]: Entering directory '/root/tizenrt/apps/examples/ntpclient_test'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/ntpclient_test'
make[3]: Entering directory '/root/tizenrt/apps/examples/mqtt_test'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/mqtt_test'
make[3]: Entering directory '/root/tizenrt/apps/examples/dtls_server'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/dtls_server'
make[3]: Entering directory '/root/tizenrt/apps/examples/kernel_sample'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/kernel_sample'
make[3]: Entering directory '/root/tizenrt/apps/examples/mpu'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/mpu'
make[3]: Entering directory '/root/tizenrt/apps/examples/libcoap_client'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/libcoap_client'
make[3]: Entering directory '/root/tizenrt/apps/examples/proc_test'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/proc_test'
make[3]: Entering directory '/root/tizenrt/apps/examples/grpc_greeter_client'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/grpc_greeter_client'
make[3]: Entering directory '/root/tizenrt/apps/examples/tls_selftest'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/tls_selftest'
make[3]: Entering directory '/root/tizenrt/apps/examples/tls_benchmark'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/tls_benchmark'
make[3]: Entering directory '/root/tizenrt/apps/examples/iotjs_startup'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/iotjs_startup'
make[3]: Entering directory '/root/tizenrt/apps/examples/sensorbd_demo'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/sensorbd_demo'
make[3]: Entering directory '/root/tizenrt/apps/examples/protobuf'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/protobuf'
make[3]: Entering directory '/root/tizenrt/apps/examples/iotivity_simpleserver'
make[3]: Nothing to be done for 'context'.
make[3]: Leaving directory '/root/tizenrt/apps/examples/iotivity_simpleserver'
make[3]: Entering directory '/root/tizenrt/apps/examples/cxxtest'
[... 1170 lines cut ...]
                 from netdb/lib_freeaddrinfo.c:28:
/root/tizenrt/os/include/net/lwip/inet.h:106:30: note: previous declaration of 'in6addr_any' was here
 extern const struct in6_addr in6addr_any;
                              ^
netdb/lib_freeaddrinfo.c: In function 'freeaddrinfo':
netdb/lib_freeaddrinfo.c:58:23: error: storage size of 'req' isn't known
  struct req_lwip_data req;
                       ^
netdb/lib_freeaddrinfo.c:67:13: error: 'FREEADDRINFO' undeclared (first use in this function)
  req.type = FREEADDRINFO;
             ^
netdb/lib_freeaddrinfo.c:67:13: note: each undeclared identifier is reported only once for each function it appears in
netdb/lib_freeaddrinfo.c:70:22: error: 'SIOCLWIP' undeclared (first use in this function)
  ret = ioctl(sockfd, SIOCLWIP, (unsigned long)&req);
                      ^
netdb/lib_freeaddrinfo.c:58:23: warning: unused variable 'req' [-Wunused-variable]
  struct req_lwip_data req;
                       ^
Makefile:133: recipe for target 'bin/lib_freeaddrinfo.o' failed
make[1]: Leaving directory '/root/tizenrt/lib/libc'
LibTargets.mk:228: recipe for target 'libc/libc.a' failed
make[1]: *** [bin/lib_freeaddrinfo.o] Error 1
make: *** [libc/libc.a] Error 2

```

zhouxinhe, 2019-05-18:
@ChiShaoJun 
Hello, the error above is a dependency problem,
you have enabled iotivity stack, but network hasn't been enabled.
Unfortunately even if you enable network, there's still some errors... 
IoTivity/SmartThings on ESP32 is not released yet, we will finish it in the near future.

sunghan-chang, 2019-05-30:
@ChiShaoJun Origianl issue was resolved and PR to use st-things on esp was raised. Let's close this issue. If you have anything on st-things, could you open another issue?

closed 2019-05-30 by sunghan-chang