Samsung/TizenRT #3249: build error on esp

sunghan-chang, 2019-05-29:
Hello.

```
make[3]: Entering directory '/home/sunghan/Work/TinyAra/TizenRT/external/esp_idf_port/soc'
Makefile:1: /home/sunghan/Work/TinyAra/TizenRT/os/.config: No such file or directory
make[3]: *** No rule to make target '/home/sunghan/Work/TinyAra/TizenRT/os/.config'.  Stop.
make[3]: Leaving directory '/home/sunghan/Work/TinyAra/TizenRT/external/esp_idf_port/soc'
Makefile:142: recipe for target 'clean' failed
make[2]: *** [clean] Error 2
make[2]: Leaving directory '/home/sunghan/Work/TinyAra/TizenRT/external/esp_idf_port'
Makefile:113: recipe for target 'esp_idf_port/_clean' failed
make[1]: *** [esp_idf_port/_clean] Error 2
```
When we execute `make clean` or `make distclean` without .config, we can meet above.
But I see no `CONFIG_XX` in Makefile of esp_idf_port/soc.
Could you let me know why you include .config there?

sunghan-chang, 2019-05-29:
@SRCN-LiQin @zhouxinhe @ChinaTizenRT @xiarihf @xixidodo @glb1013 Could you check this?

ChinaTizenRT, 2019-05-30:
tks for your notice. this place does not need include .config, if adding,  had better using -include instead of include as .config is file created, not original one.

below PR is opened for that.
https://github.com/Samsung/TizenRT/pull/3250

and I checked external/esp_idf_port/spi_flash also has similar issue,
but PR https://github.com/Samsung/TizenRT/pull/3060 
discard the spi_flash itself's makefile, using arch/xtensa/src/chip/Makefile
so this issue not exisit.

sunghan-chang, 2019-05-30:
@ChinaTizenRT I did both. Thank you for resolving.

sunghan-chang, 2019-05-30:
make[3]: Entering directory '/home/sunghan/Work/TinyAra/Forked_TizenRT/external/esp_idf_port/esp32'
Makefile:1: /home/sunghan/Work/TinyAra/Forked_TizenRT/os/.config: No such file or directory
make[3]: *** No rule to make target '/home/sunghan/Work/TinyAra/Forked_TizenRT/os/.config'.  Stop.
make[3]: Leaving directory '/home/sunghan/Work/TinyAra/Forked_TizenRT/external/esp_idf_port/esp32'
Makefile:142: recipe for target 'clean' failed
make[2]: *** [clean] Error 2
make[2]: Leaving directory '/home/sunghan/Work/TinyAra/Forked_TizenRT/external/esp_idf_port'
Makefile:114: recipe for target 'esp_idf_port/_distclean' failed
make[1]: *** [esp_idf_port/_distclean] Error 2

@ChinaTizenRT There is the same issue on another folders. Could you check all of esp_idf_port?

ChinaTizenRT, 2019-05-31:
This PR is for the issue.  I also verify it by run "make clean" withou .config file
no error shows up in esp_idf_port.

https://github.com/Samsung/TizenRT/pull/3267

closed 2019-05-31 by sunghan-chang