Samsung/TizenRT #3258: This should NOT be fixed like this. It depends on CONFIG_FLASH_PART_SIZE="256,256,4096. Otherwise user binary will be flashed based on CONFIG_FLASH_PART_SIZE="256  and this fixed value will be  used in Code. Lot of trouble. Either fix CONFIG_FLASH_PART_SIZE="256,  (make it NOT configurable) and use fixed value or calculate the start address from partititon and use it.

btheosam, 2019-05-30:
This should NOT be fixed like this. It depends on CONFIG_FLASH_PART_SIZE="256,256,4096. Otherwise user binary will be flashed based on CONFIG_FLASH_PART_SIZE="256  and this fixed value will be  used in Code. Lot of trouble. Either fix CONFIG_FLASH_PART_SIZE="256,  (make it NOT configurable) and use fixed value or calculate the start address from partititon and use it.

_Originally posted by @btheosam in https://github.com/Samsung/TizenRT/pull/3244_

sunghan-chang, 2019-05-30:
@btheosam I agree your opinion. We already get a direction to do that. First step is #3223 . Next will be what you say.

sunghan-chang, 2020-01-07:
@sijupunnnoose @sangam-swami @thapav Could you check this?
As Benix said, we could set the value automatically. We don't want to set several points for the same value. If we apply auto-setting or auto-generate, we need to update the doc for porting protected build.

sunghan-chang, 2020-01-15:
@thapav Could you let me know the status of this request?

sunghan-chang, 2020-02-07:
@btheosam @kishore-sn pushed the change, PR #4183 . Thank you.

closed 2020-02-07 by sunghan-chang