Samsung/TizenRT #3297: tc_clock_clock_abstime2ticks fails

glb1013, 2019-06-04:
The testcase is changed to use the virtual driver.
Old version clock_abstime2ticks(CLOCK_REALTIME, &base_time, &base_tick) will return the OK and the base_tick holds the value of base_time and it is -1. So it passes.
But in the virtual driver, the return value -1 of ioctl() has two cases: the first one is the clock_abstime2ticks() return value is not OK, the second one is the clock_abstime2ticks() returns OK but the ticks is -1. In this testcase, it fails because the base_time is before current_time, clock_abstime2ticks() always returns OK with tick value is -1.

glb1013, 2019-06-04:
[tc_clock_clock_abstime2ticks] FAIL [Line : 181] clock_abstime2ticks : Values (b
ase_tick == 0xffffffff) and (-1 == 0xffffffff) are equal

sunghan-chang, 2019-06-04:
@YiHyunJin Could you check this issue?

YiHyunJin, 2019-06-04:
@sunghan-chang 
yes. i will check.

sunghan-chang, 2019-06-12:
@glb1013 @YiHyunJin modified.
1. Near end of sec, it always can be failed with current testcase.
2. The tick numbers of current time does not make sense.

Could you test it on esp? Because it is not reproducible for us.

glb1013, 2019-06-12:
@YiHyunJin @sunghan-chang
In the  older version of testcase(before commit 61097b5efbae171010c0e33be951efebf278b57c committed on 15 Jan, 2019), the base_time is not increased by 1. I guess it tests the second failure branch of clock_abstime2ticks() which return OK but the base_tick value is -1, and it means the input abstime is before currtime (it means clock_abstime2ticks() starts time).
So, now this version does not cover the second failure branch.

sunghan-chang, 2019-06-14:
@glb1013 Ok that means you run another application with testcase. Our testcase does not concern running with multiple application. Without another running application, it can not be past.
If it has a problem only on it, we can use big offsets at base_time and comparission_time.
Anyway, is it reproducible for you? As I said before, it isn't for us.

sunghan-chang, 2019-06-17:
@glb1013 The purpose of clock_abstime2ticks() is getting system time ticks from absolute timespec delay. Previous testcase is not appropriate I think. So we modified it as shown below:
1. getting ticks of 1 sec delay
2. getting ticks of 2 sec delay
3. calculating difference between 2nd and 1st
4. check whether 3rd is ticks of 1 sec or not.
With new mechanism, we don't need to get -1 as ticks.

And, commit 61097b5 was wrong. First clock_abstime2ticks can return -1 but it treated as failure.

closed 2019-06-18 by glb1013