Samsung/TizenRT #3541: Build error on esp32 board as network code modifcation.

ChinaTizenRT, 2019-07-10:
There is build issue for esp32 board, it's introuduced by network code modifcation.
Request network owner to help fix this issue.

build command:
1) ./tools/configure.sh  esp_wrover_kit/tash_wlan
2) make 
=====================================================
error logs
netdev/netdev_dhcpd.c: In function 'netdev_dhcp_server_status':
netdev/netdev_dhcpd.c:112:15: error: 'struct netif' has no member named 'dhcps_pcb'
  if (cur_netif->dhcps_pcb == NULL) {
               ^
netdev/netdev_dhcpd.c: In function 'netdev_dhcp_server_stop':
netdev/netdev_dhcpd.c:158:15: error: 'struct netif' has no member named 'dhcps_pcb'
  if (cur_netif->dhcps_pcb == NULL) {
               ^
netdev/netdev_dhcpd.c: In function 'netdev_dhcp_server_status':
netdev/netdev_dhcpd.c:119:1: warning: control reaches end of non-void function [-Wreturn-type]
 }

@sh.byeon@samsung.com

SeonghoByeon, 2019-07-22:
Hi @ChinaTizenRT, the error from DHCP is solved by https://github.com/Samsung/TizenRT/pull/3588. Please refer to it. However, still there is another problem related with partitions. To handle it, I think you need to discuss it with @jeongchanKim and @YiHyunJin. Thanks.

sunghan-chang, 2019-08-01:
@ChinaTizenRT @SeonghoByeon modified what you said. If you have an issue, please let us know.

closed 2019-08-01 by sunghan-chang