Samsung/TizenRT #3591: can't load the emulator from the IDE tizen studio.

ping996, 2019-07-22:
I can't load the emulator from the tizen studio when click the tools->emulator managers.
error: could not create the java virtual machine.
I use the open JDK 10 and set the JAVA_HOME and path OK by input java -version in cmd.
what's problems?

sunghan-chang, 2019-07-24:
@ping996 Sorry for late reply. Let us check it first.

hasw7569, 2019-07-24:
@ping996
Are you using Tizen RT IDE ?
To check this issue, I installed the Tizen RT IDE.
However, I did not find the emulator manager in the tools menu.

If you using Tizen RT IDE, what board are you using ?

In addition, the following link provides a debugging guide. Have you followed this guide?
https://developer.tizen.org/development/tizen-studio/rt-ide/getting-started/debugging-project

ping996, 2019-07-26:
@hasw7569  I installed the tizen studio, not the RT version.  now It's working now by changed some setting of my window 10 system.

sunghan-chang, 2019-08-01:
@ping996 Actually we have separated RT IDE in tizen studio. Please refer @hasw7569 's link and let us know when you have an issue.

closed 2019-08-01 by sunghan-chang