Samsung/TizenRT #3850: Review result of /docs

ck11a, 2019-10-01:
@sangwon03 

Hello, SangWon

I'm Sang-A Lee from SE team. 
I have completed the documentation review of TizenRT's docs folder.
After the GBM release, please modify when you can afford, and attach the modified PR to the answer.

Thank you. 

[README.md]

\# 1. line 11
-Original
> [Environment Setup](#environment-setup)  
> [How to Build](#how-to-build)  
> [Supported Board / Emulator](#supported-board--emulator)  

-Suggestion
- [Environment Setup](#environment-setup)  
- [How to Build](#how-to-build)  
- [Supported Board / Emulator](#supported-board--emulator)  

-Review comment
I recommend using Unordered Lists rather than BlockQuote for the list of contents. Please refer to style guide. https://github.com/Samsung/tizen-docs/blob/master/styleguide/template-guide.md

\# 2. line 24
-Original 
Especially For Windows environment, 
-Suggestion
Especially for Windows environment,
-Review comment
There is a capital letter in the middle of the sentence.

\# 3. line 24
-Original
Especially For Windows environment, configuring CRLF is required as below.
```
git config --global core.autocrlf input
```
-Suggestion
-Review comment
I recommend that the CRF content be located at the end of 2. Getting TizenRT source code step. Since CRF configuration affects commit, it is most relevant to step 2.

\# 4. line 31
-Original
To install OS specific Docker engines, see https://docs.docker.com/install/linux/docker-ce/ubuntu/.
-Suggestion
To install OS specific Docker engines, see https://docs.docker.com/install/.
-Review comment
You should write the install page url for all OS, not the url for specific OS such as Linux.

\# 5. line 29
-Original 
### 1. Installing Docker
### 2. Getting TizenRT source code
-Suggestion
### 1. Installing Docker
### 2. Pulling TizenRT Docker Image
### 3. Getting TizenRT source code
-Review comment
I think a step to pull TizenRT Docker Image should be added between steps 1 and 2.

\# 6. line 41
-Original 
To contribute in this community, you need to clone your forked private repository instead.
-Suggestion
To contribute in this community, you need to clone the private repository that forked from TizenRT project.
-Review comment
It is better to specify the target project for fork.
[... 2347 lines cut ...]
SmartThings&trade; uses OpenSSL for security. To ensure that only trusted developers and manufactures can develop devices that communicate with the SmartThings Cloud, you must get certificates for devices to use the SmartThings&trade; service.

## Contents
- [Using a Certificate Manager](#using-a-certificate-manager) 
- [Creating a Certificate](#creating-a-certificate)  
- [Importing a Certificate](#importing-a-certificate)  
- [Activating a Certificate](#activating-a-certificate)  

## Using a Certificate Manager
In the TizenRT IDE, you can perform various certificate-related tasks, such as creation, removal, import, and activation, by using the **SmartThings Certificate Manager**.

[image]

The Certificate Manager is launched automatically if no certificate is set as active when you try to flash a project. To launch the Certificate Manager manually, go to **Tools &gt; ST Certificate Manager** in the Tizen Studio for RT menu.

-Review comment
I recommend changing to a layout like Suggetion.

\# 3. line 14
-Original
To launch the Certificate Manager manually, go to **Tools &gt; ST Certificate Manager** in the Tizen Studio for RT menu.
-Suggestion
To launch the Certificate Manager manually, select **Tools &gt; ST Certificate Manager** in the Tizen Studio for RT menu.
-Review comment
I recommend changing it like Suggestion.

dr-venkman, 2019-11-21:
@ck11a 
Thank you for your comments, we have addressed most of them in pull request #4028 . Do let us know if you have further comments.

sunghan-chang, 2019-12-04:
@ck11a We merged your  comments. When you have any update, please open the issue or create the PR. Thank you for supporting.

closed 2019-12-04 by sunghan-chang