Samsung/TizenRT #3986: TizenRT security module build error issue

ChinaTizenRT, 2019-10-26:
File: framework/src/security/security_ss.c
Line: 109, 129
cause:  logging function may be null, leads to unused variable.
error log:
src/security/security_ss.c: In function 'ss_get_size_secure_storage':
src/security/security_ss.c:109:23: error: unused variable 'ctx' [-Werror=unused-variable]
  struct security_ctx *ctx = (struct security_ctx *)hnd;
                       ^~~
src/security/security_ss.c: In function 'ss_get_list_secure_storage':
src/security/security_ss.c:129:23: error: unused variable 'ctx' [-Werror=unused-variable]
  struct security_ctx *ctx = (struct security_ctx *)hnd;
                       ^~~

pillip8282, 2019-11-12:
Thanks for your report. the patch is uploaded at https://github.com/Samsung/TizenRT/pull/4020

sunghan-chang, 2019-11-12:
@ChinaTizenRT #4020 was merged. If you have a problem, please (re) open an issue.

closed 2019-11-12 by sunghan-chang