Samsung/TizenRT #4013: How to import and build on TizenRT Studio

HotCodeworker, 2019-11-09:
I installed Tizen Studio 3.5 with RT IDE on Ubuntu 64bit as next link:
[https://developer.tizen.org/development/tizen-studio/download]

But, I can manage a project because the GUI menus of RT IDE installed on my Ubuntu are different from such as next link: [https://developer.tizen.org/development/tizen-studio/rt-ide/getting-started/managing-project]

In details, the menus of the current version of RT IDE are missing from as above link.

Please let us know how to use(import or build) the sources of TizenRT github's repository currently.

HotCodeworker, 2019-11-09:
In addition, I installed openJDK such as follows: [https://developer.tizen.org/development/articles/openjdk-and-openjfx-installation-guide](https://developer.tizen.org/development/articles/openjdk-and-openjfx-installation-guide)

HotCodeworker, 2019-11-11:
When using oracle JDK, this issue could be solved. This issue might happen when using openJDK.

manishtr, 2019-11-12:
Hi,
Please install RT ide from http://10.113.138.168/packages/tizen_studio/releases/Tizen_Studio_3.3/ , you can install RT_Tizen_Studio_3.3_ubuntu-32.bin or RT_Tizen_Studio_3.3_ubuntu-64.bin.
As RT ide is not supported from 3.5 hence the public link does not have the installer.
RT ide does not support OpenJDK or Java 12.
You need to use java 8.
Regards,
Manish

sunghan-chang, 2019-11-12:
@HotCodeworker I am sorry for inconvenient. We will add the sentence to say Java 8 in guidance.

sunghan-chang, 2019-11-12:
@HotCodeworker Let's close this and manage this issue in #4014 .

closed 2019-11-12 by sunghan-chang