Samsung/TizenRT #4014: Can't launch TizenRT studio!!!

HotCodeworker, 2019-11-09:
I just installed the TizenRT IDE using Tizen Studio 3.5 with RT IDE installer:[https://developer.tizen.org/development/tizen-studio/download](https://developer.tizen.org/development/tizen-studio/download)

However, I can't launch the RT IDE such as below attached picture:
[image]

Let us know how to fix it and thanks in advance.

HotCodeworker, 2019-11-09:
In addition, I installed openJDK such as follows: [https://developer.tizen.org/development/articles/openjdk-and-openjfx-installation-guide](https://developer.tizen.org/development/articles/openjdk-and-openjfx-installation-guide)

HotCodeworker, 2019-11-11:
When using oracle JDK, this issue could be solved. This issue might happen when using openJDK.

manishtr, 2019-11-12:
Hi,
RT ide does not support OpenJDK or Java 12.
You need to use java 8.
Please try to install and use a new workspace.
Regards,
Manish

sunghan-chang, 2019-11-12:
@HotCodeworker I am sorry for inconvenient. We will add the sentence to say Java 8 in guidance.

manishtr, 2019-11-12:
I have raised the PR for Documentation update at https://github.com/Samsung/tizen-docs/pull/937

sunghan-chang, 2019-11-21:
@HotCodeworker In prerequisites of Tizen Studio, we added the **NOTE** to say above.
https://developer.tizen.org/development/tizen-studio/download/prerequisites
>Note
>For RT IDE use Oracle Java Development Kit (JDK) 8 only.

If you have any problem to use RT IDE, please let us know.

closed 2019-11-21 by sunghan-chang