Samsung/TizenRT #4015: Support of RT IED on esp32 board?

HotCodeworker, 2019-11-11:
I'm using on ESP32 board and RT IDE.
On RT IDE, I used "Batch build project".
However, I can not used the function like "Application Manager" and "Kernel config" because RT IDE may not enable the functions.

Originally, the functions should be enabled as next links:[https://developer.tizen.org/development/tizen-studio/rt-ide/getting-started/managing-project](https://developer.tizen.org/development/tizen-studio/rt-ide/getting-started/managing-project)

then please let me know how and what.

zhouxinhe, 2019-11-12:
Hello HotCodeworker

To use `Application Manager` and `Kernel Config` function:
1. Select the project in the Project Explorer view.
2. Click the arrow next to the Build TizenRT Project icon.
  Then you can see `Application Manager` and `Kernel Config` items  in the drop-down menu.

I tried to select `Application Manager`, but no dialog appeared (I don't know why...).
Fortunately, we can manage application in `Kernel Config` -> Application Configuration -> ...

sunghan-chang, 2019-11-12:
> I tried to select Application Manager, but no dialog appeared (I don't know why...).

Let me check it.

HotCodeworker, 2019-11-12:
In addition, my issue are only about ESP32_DevKit board.
More issue with ESP32_DevKit board is also that I couldn't debug with this ESP32 board on RT IDE beacuse RT IDE poped-up like that "there is no debug.xml".

sunghan-chang, 2019-11-13:
@manishtr When we check `make applist` or `dbuild.sh applist`, it creates .appSpec file. Could you check why it does not work with Application Manager?

dkdwivedi, 2019-11-15:
launching Application manager creates same .appSpec file as generated by `make applist`

sunghan-chang, 2019-11-15:
@dkdwivedi Thank you for helping.
@HotCodeworker  This is resolved with https://github.com/Samsung/TizenRT/commit/572d61278e226b5e32e3a8cdd244fd9fd1bd0ce8. After checking Application Manager on IDE, let you know at next Monday.

sunghan-chang, 2019-11-18:
@HotCodeworker
1. Application Manager is working now with latest commit.
2. As you said, debug feature is under implementation. Let's talk it in #4029 .

Currently kerne config and application manager are working well. Let's close this and when you have a problem, let me know. Thank you for the issue.

closed 2019-11-18 by sunghan-chang