Samsung/TizenRT #4016: Support on ESP32-DevKitC board!

HotCodeworker, 2019-11-11:
I was compiling on ESP32-DevKitC board.
Then when we selected download option and performed it using "./dbuild downlaod", the error occured like this: "there is no esptool.py under the /home/hotcoder/git/TizenRT/build/tools/esp32/esptool_py/esptool"

I think the file, esptool.py was NOT created after compiling and building using "ESP32-DevKitC board" configuration on terminal invoked by "./dbuild.sh".

Then please let me know why and thanks in advance.

xiarihf, 2019-11-12:
Dear HotCodeworker, the esptool.py is the submodule of TizenRT due to the license, so please get the submodules first by using following git command.
            git submodule update --init --recursive 
We will add a checking operation before downloading images soon.
Thanks for your feedback.

HotCodeworker, 2019-11-12:
Thanks for your reply and I've solve it.

closed 2019-11-13 by sunghan-chang