Samsung/TizenRT #4019: booting error - on ESP32-D evKitC board

HotCodeworker, 2019-11-11:
I was working on ESP32-DevKitC board right after downloading esptool.py apart from git([https://github.com/espressif/esptool/blob/master/esptool.py](https://github.com/espressif/esptool/blob/master/esptool.py)) because the error as "there is no esptool.py" occured.

So like this method, I succeeded flashing image but the board couldn't boot normally with the following logs:

--------------------------------------------------------------------------------------------------------
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5752
load:0x40078000,len:9020
load:0x40080000,len:6064
entry 0x40080330
I (29) boot: ESP-IDF v3.1-dirty 2nd stage bootloader
I (29) boot: compile time 18:12:49
I (29) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 2MB
E (46) flash_parts: partition 2 invalid - offset 0x10000 size 0x280000 exceeds flash chip size 0x200000
E (56) boot: Failed to verify partition table
E (61) boot: load partition table error!
user code done
------------------------------------------------------------------------------------------------------------

please check this ESP32 board with TizenRT.

ChinaTizenRT, 2019-11-12:
@HotCodeworker HotCodeworker ,   I pull TizenRT latest code
(id: 72f93d63e57fcedd80c51b9adef57b32eb5d19db), booting is okay,
pls check below steps.
1)  use esp32_DevKitC/hello_with_tash defconfig.
2)  run "make download ALL" to flash image for the first time.
if above steps isn't okay for you, before flashing, pls erase the entire flash via command
3) make download erase_flash. then switch to step 2.
Wish you good luck.

ChinaTizenRT, 2019-11-12:
from the error log 
E (46) flash_parts: partition 2 invalid - offset 0x10000 size 0x280000 exceeds flash chip size 0x200000
seems that image size is bigger than your chip size 2MB.

as i know.  there are 4MB SPI FLASH in esp32_DevKitC, pls check your defconfig like below
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y

HotCodeworker, 2019-11-12:
Thanks for response and I've solve it,

sunghan-chang, 2019-11-13:
@HotCodeworker Thank you for confirmation. Let's close this.
@ChinaTizenRT Thank you for checking.

closed 2019-11-13 by sunghan-chang