Samsung/TizenRT #4029: Debug problem on RT IDE with ESP32_DevKit board?

HotCodeworker, 2019-11-12:
I'll let you know my issue are only about ESP32_DevKit board.
The issue with ESP32_DevKit board is that I couldn't debug with this ESP32 board on RT IDE because RT IDE popped-up like that "there is no debug.xml".

zhouxinhe, 2019-11-13:
Dear HotCodeworker

Debugging with this ESP32 board on RT IDE is not supported now.
We plan to enable it in near future. Then I will notify you here when it is ready. 
Sorry for any inconvenience.

sunghan-chang, 2019-11-13:
@zhouxinhe Could you let us know the ETA?

zhouxinhe, 2019-11-13:
> @zhouxinhe Could you let us know the ETA?

Maybe until next weekend, is that ok?

sunghan-chang, 2019-11-13:
@HotCodeworker Could you confirm the our schedule?

HotCodeworker, 2019-11-13:
When it is available the next week, I'll try the debug function again.

zhouxinhe, 2019-11-19:
@HotCodeworker 
I just raised PR #4042 to fix this issue.
Please read the PR description before you trying it.