Samsung/TizenRT #4030: The problem with openinig terminal on RT IDE with ESP32_DevKit board.

HotCodeworker, 2019-11-12:
This issue is the problem with opening terminal on RT IDE and this may be not the issue.
When I tried to open terminal on "/dev/ttyUSB0" on Ubuntu PC which usb-connected with ESP32 board, I should do first like this command, "sudo chmod 777 /dev/ttyUSB0" on the bash terminal apartly from on RT IDE.

My idea is that it would be nice if there would not need more user's inconvience like above.

sunghan-chang, 2019-11-13:
@hasw7569 Could you check this?

hasw7569, 2019-11-13:
@HotCodeworker

hasw7569, 2019-11-13:
In my development environment, the terminal works fine without sudo permission or chmod command.

hasw7569, 2019-11-13:
On my Ubuntu PC the terminal works fine without sudo permission or chmod command.
I think this problem occurred in a special environment.
Did you get a Permission denied message when you tried to open a terminal?
If not, please let me know what kind of error occurred.

HotCodeworker, 2019-11-13:
I got a Permission denied message when you tried to open a terminal without sudo permission or chmod command. So I solve it with sudo and chmod command apart not from on RT IDE but from terminal. However, it's very inconvient for us, I think.

hasw7569, 2019-11-13:
What is the first permission on ttyUSB0 before you enter the chmod command?

hasw7569, 2019-11-13:
I think that this link will help you

[https://github.com/Samsung/TizenRT/blob/master/build/configs/artik053/README.md#environment-set-up]

Environment Set-up -> Environment Set-up

HotCodeworker, 2019-11-13:
@hasw7569 Thanks for your reply, it'll help for me.

sunghan-chang, 2019-11-14:
@HotCodeworker Thank you for good suggestion. Let us check to register USB rule automatically.

sunghan-chang, 2019-11-27:
@HotCodeworker We added the functionality that registers the USB rule in dbuild script. Please find #4063 . You can use it on IDE as well as on CLI. It adds the `USBrule` option in download step. When you select it, it checks whether USB rule is registered or not, if it isn't, it registers automatically.
Because some developer does not like changing his env by us, we provide it as an option.
If you have any other suggestion, please let us know.

closed 2019-12-04 by sunghan-chang