Samsung/TizenRT #4031: No memory information after building project on ESP32_DevKitC board.

HotCodeworker, 2019-11-12:
The log are as follows after build project.

-------------------------------------------------------------------------------------
....
make[1]: Leaving directory '/root/tizenrt/framework'
make[1]: Entering directory '/root/tizenrt/os/arch/xtensa/src'
make[2]: Entering directory '/root/tizenrt/os/board/esp32_devkit/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/tizenrt/os/board/esp32_devkit/src'
LD: tinyara
make[1]: Leaving directory '/root/tizenrt/os/arch/xtensa/src'
CP: tinyara.elf.bin
Start the Board Specific Work for Binary
######################################
##          Library Sizes           ##
######################################
	.data 	.bss 	.text 	 Total
[STATUS] SUCCEED

--------------------------------------------------------------------------------------

Then there is no memory information on above log with "Libray Sizes" label.
This issue happened on both RT IDE and CLI on terminal.

sunghan-chang, 2019-11-13:
@xixidodo Could you assign the person to check this?
@HotCodeworker Currently we support showing libraries size for GNU toolchain. But esp32's toolchain is not GNU so that we need to check it.

xixidodo, 2019-11-13:
@HotCodeworker @sunghan-chang 
Currently,  @ChinaTizenRT  is checking this now,  xtensa toolchain is not same, we need to check why stat is not retrieved.

ChinaTizenRT, 2019-11-14:
@HotCodeworker, internally job is done for this feature, after review, patch will be released.

sunghan-chang, 2019-11-21:
@HotCodeworker PR #4048 , commit 2f58ee7cd74970f7a4ccbb7fa6869e5a8168219a was merged. With latest codes, it prints libraries information. If you have any issue on it, please let us know.

closed 2019-11-21 by sunghan-chang