Samsung/TizenRT #4032: No flash functionality on RT IDE with ESP32_DevKitC

HotCodeworker, 2019-11-12:
The flash functionality on RT IDE with ESP32 board was not enabled and disabled so I couldn't use flash function with SP32_DevKitC board. Of course, I could use this function using "dbuild.sh" on terminal. This issue is only on RT IDE.

sunghan-chang, 2019-11-13:
@HotCodeworker Because of missing **.flashSpec.xml* under *https://github.com/Samsung/TizenRT/tree/master/build/configs/esp32_DevKitC*, it did not work. But yesterday we merged 5d53bc45c7e93751444a27596d107df8f3b70b4d. Could you check it with latest?

zhouxinhe, 2019-11-13:
@sunghan-chang @HotCodeworker 
yes, I fixed it in PR #4024, but please wait PR #4023 (#4024 depends on it).

sunghan-chang, 2019-11-18:
@HotCodeworker Both #4023 and #4024 were merged. 
@zhouxinhe Could you confirm the functionality of flash using IDE with latest?

zhouxinhe, 2019-11-19:
@HotCodeworker @sunghan-chang 
Dear, I tested the functionality of flash using IDE with latest code, it's working fine.

sunghan-chang, 2019-11-19:
@zhouxinhe Thank you for confirmation.
@HotCodeworker Flashing is working with latest code on IDE. Let's close this issue now. If you have any issue again, please kindly reopen the issue.

closed 2019-11-19 by sunghan-chang