Samsung/TizenRT #4033: The problem with the build option on ESP32_DevKitC!

HotCodeworker, 2019-11-12:
On RT IDE, the error log with performing "build project" are follows after I selected the build option as both "tash_wlan" and "tc".

----------------------------------------------------------
Kenel config execution failed.
Please check your configuration again.
----------------------------------------------------------

But I could build with above build options apartly on terminal.

sunghan-chang, 2019-11-13:
@HotCodeworker Let me check.

sunghan-chang, 2019-11-14:
@HotCodeworker There are two issues.
1. Not configured for ESP32_DevKitC
  Because of incorrect os/dbuild.sh, that board is not configured.
  We check the modification and it works well. We will make a PR soon.
2. IDE issue
  There are three ways to build in IDE, ICON, "build project" menu at project explorer and "build project" at menu.
  There is no problem with ICON. It works well.
  But both "build project" requires manual configuration first. Before executing it, you should configure board and configuration first. Then it works. 
  As you know, it is inconvenient. We will change it as ICON does.

sunghan-chang, 2019-11-14:
1st issue was resolved. #4046

sunghan-chang, 2019-11-19:
2nd will be patched soon on IDE side. (This is an issue on usage. As of now, there is no problem to build with any way.)

sunghan-chang, 2019-11-25:
@HotCodeworker This issue was resolved. Let me summarize this issue.
- Issue : build break like above
- Reason
    1. Configuration is not done because of the error in dbuild script. (ESP32_DevKitC is not recognized). #4046 in TizenRT source code have resolved.
    2. All of build methods, using icon (shortcut), build project at project explorer window and build project at menu, does not give the same usages. Only icon goes to configuration window automatically when configuration is not done. This was resolved in RT IDE. You could find the change at next release of RT IDE, it might be end of this month.

When new version of RT IDE is released, I will close this issue.

sunghan-chang, 2019-12-16:
@HotCodeworker We've checked the functionality with 3.6 Tizen Studio. It works well. If you have any question, please let us know.

closed 2019-12-16 by sunghan-chang