Samsung/TizenRT #4073: How to install or update tizenRT IDE ?

HotCodeworker, 2019-11-25:
I'd like to use the latest tizenRT IDE with latest codes on ESP32 board. please let me know how to update existing old RT IDE with the latest.

sunghan-chang, 2019-11-26:
@manishtr  Could you let us know how to do it?

manishtr, 2019-11-26:
Please follow the below steps,
Step 1: Launch the package manager from ~/tizen-studio/package-manager/package-manager.bin
Step 2: Click on the setting icon on the top right (middle button amont ? O and i)
Step 3: Paste the following in the Package Repository  http://10.113.138.168/srib-packages
Step 4: Select Distribution as develop and click OK.
Step 5: Install RT IDE. This should install the updated RT IDE.

manishtr, 2019-11-26:
Latest RT Ide version is 3.5, it can be taken from https://developer.tizen.org/development/tizen-studio/download

sunghan-chang, 2019-11-26:
@HotCodeworker Here is the guidance of updating Tizen Studio.
https://developer.tizen.org/development/tizen-studio/download/installing-tizen-studio/updating-tizen-studio
It might be an answer about your query.

HotCodeworker, 2019-11-28:
@sunghan-chang Thanks for your reply and it was helpful then I just checked if there is no update of the RT IDE than the current on package-manager. In addition, the current version is just 3.5.

sunghan-chang, 2019-11-29:
@HotCodeworker Yeah, v3.5 is current version of IDE. With next version, build issue which is mentioned in #4033  will be resolved.
You can check #4032 , #4031 , #4019 , #4015 issues with latest TizenRT source codes, not change in IDE side.
We made a NOTE to notice the version of JAVA in prerequistes page of Tizen Studio for #4014  .

sunghan-chang, 2019-11-29:
@HotCodeworker Before new IDE, please configure first and then build TizenRT or use shotcut icon.
Because of missing build configuration, you met #4033 issue.

sunghan-chang, 2019-11-29:
@HotCodeworker Sorry I gave you wrong information. v3.5 is right what I said. Let me check it first.

sunghan-chang, 2019-12-02:
Modified version is not released yet. It could be end of this week or next week. v3.5 was released at Oct. 31th. Our change was merged at Nov. (I confused Oct and Nov) When new one is released, I will let you know.

sunghan-chang, 2019-12-20:
@HotCodeworker As I left at another issue, Tizen Studio v3.6 was released. It works well what you left issues except debugging on ESP32. If you don't have any problem, let's close this issue as well.
cf) debugging issue on ESP32 could be treated on #4029 .

closed 2019-12-30 by sunghan-chang