Samsung/TizenRT #4219: Wi-fi not work on Artik 053

RedCarrottt, 2020-03-04:
Hello, I want to use Wi-fi on my Artik 053(SIP-0P5WRS302).

I built Tizen RT of version fb5049d72ac388753935cd3a8f8e0e6c1c047716 with following build config (using `os/dbuild.sh menu`).
* board: artik053
* configuration: nettest

However, after downloading the Tizen RT on my Artik 053, it fails to connect to my access point (without any passphrase) and it results in rebooting.
* SSID: redc
* SECURITY: open (no passphrase)

I can get following messages.
How can I resolve this issue?
```
TASH>>wm_test start
TASH>>wifi manager test!!
 T8 wait func signal
-->wm_start
[WM] T9 _handle_request:1285 state(0) evt(0)
[WM] T9 _handler_on_uninitialized_state:839 state(0) evt(0)
Starting supplicant in foreground...
198.310567: Successfully initialized wpa_supplicant
201.404057: wl1: callling L2_packet_init:
201.404057: wl1: Own MAC address: 28:6d:97:40:14:91
<--wm_start
 T9 send func signal
TASH>>wm_test join redc open
TASH>>wifi manager test!!
 T14 wait func signal
-->wm_connect
====================================
Connecting AP Info
------------------------------------
SSID: redc
SECURITY: open
====================================
[WM] T15 _handle_request:1285 state(1) evt(4)
[WM] T15 _handler_on_disconnected_state:882 state(1) evt(4)
[WM] T15        _wifimgr_connect_ap:676
[WM] T15        _wifimgr_save_connected_config:664
 wait join done
 T15 wait signal

U-Boot 2017.01-g59977f7 (Nov 23 2017 - 16:09:49 +0900)

CPU:   Exynos200 @ 320 MHz
Model: ARTIK-053 based on Exynos T20
DRAM:  946 KiB
WARNING: Caches not enabled
BL1 released at 2017-3-13 15:00
SSS released at 2017-09-12
WLAN released at 2017-11-30
Flash: 8 MiB
*** Warning - bad CRC, using default environment

In:    serial@80180000
Out:   serial@80180000
Err:   serial@80180000
Hit any key to stop autoboot:  0
gpio: pin gpg16 (gpio 46) value is 1
## Starting application at 0x040C8020 ...
s5j_sflash_init: FLASH Quad Enabled
i2c_uioregister: Registering /dev/i2c-0
i2c_uioregister: Registering /dev/i2c-1
up_spiinitialize: Prepare SPI0 for Master operation
up_spiinitialize: Prepare SPI1 for Master operation
up_spiinitialize: Prepare SPI2 for Master operation
up_spiinitialize: Prepare SPI3 for Master operation
System Information:
        Version: 3.0
        Commit Hash: 46f5dbe3acad35186339f3583dd5cc94b5656a03
        Build User: root@545a532dc414
        Build Time: 2020-03-04 09:49:44
        System Time: 07 Feb 2106, 06:28:15 [s] UTC Hardware RTC Support
TASH>>Hello, World!!
```

RedCarrottt, 2020-03-18:
When I connected my Artik 053 to USB 2.0 port, this error occured.
**After connecting it to USB 3.0 port, the error disappears and it works well.**

I guess that the power supply through USB 2.0 port was not sufficient.

However, it is problem that any error message or caution message does not appear in this situation.
**Is there any schedule to implement printing caution message on the low-power situation?**

sunghan-chang, 2020-03-18:
@RedCarrottt I am sorry but ARTIK053 chipset vendor announced not to support and it is difficult to get the register information for that. There is no plan to do something for ARTIK053.

RedCarrottt, 2020-03-18:
@sunghan-chang I see.
Then, what is the main target device of Tizen RT nowadays?
I want to use IoT.js and IoTivity with Tizen RT.
Is there any proper target device that I can use?

Thank you.

sunghan-chang, 2020-03-24:
@RedCarrottt I am so sorry for late reply but I was in vacation. Recently we used I. MX RT 10X0 boards. When you configure what you want, please feel free to contact us.

RedCarrottt, 2020-03-24:
@sunghan-chang Okay I see. Thank you.

Since my issue is resolved, I close this issue.

closed 2020-03-24 by RedCarrottt