Samsung/TizenRT #4325: how to capture playback sound use tinyalsa?

jerryyyq, 2020-06-23:
I want capture playback sound on ubuntu and android.
I call pcm_open()  and  pcm_readi()  and I try:
mixer_get_ctl_by_name(mixer, "Loopback Mixing");
mixer_ctl_set_enum_by_string(ctl, "Enabled");

But, allways record wav from MIC, not from playing sound.
How to do I can?
Thank you very much.

-------------- My sound card Device Info: ------------------------
```
yangyuqi@yangyuqi-T450:~/sound$ tinymix -D 1 contents
Number of controls: 30
ctl	type	num	name                                    value
0	INT	2	Speaker Playback Volume                 0, 0 (range 0->87)
1	BOOL	2	Speaker Playback Switch                 Off, Off
2	INT	2	Headphone Playback Volume               87, 87 (range 0->87)
3	BOOL	2	Headphone Playback Switch               On, On
4	BOOL	2	Headphone Playback Switch               On, On
5	ENUM	1	Loopback Mixing                         Disabled, > Enabled, 
6	INT	2	Dock Mic Playback Volume                0, 0 (range 0->31)
7	BOOL	2	Dock Mic Playback Switch                Off, Off
8	INT	2	Mic Playback Volume                     0, 0 (range 0->31)
9	BOOL	2	Mic Playback Switch                     Off, Off
10	ENUM	1	Auto-Mute Mode                          Disabled, > Enabled, 
11	INT	2	Capture Volume                          37, 37 (range 0->63)
12	BOOL	2	Capture Switch                          On, On
13	INT	2	Dock Mic Boost Volume                   0, 0 (range 0->3)
14	INT	2	Mic Boost Volume                        0, 0 (range 0->3)
15	INT	2	Internal Mic Boost Volume               0, 0 (range 0->1)
16	INT	2	Beep Playback Volume                    0, 0 (range 0->31)
17	BOOL	2	Beep Playback Switch                    Off, Off
18	ENUM	1	Mic Mute-LED Mode                       On, Off, Follow Capture, > Follow Mute, 
19	INT	1	Master Playback Volume                  66 (range 0->87)
20	BOOL	1	Master Playback Switch                  On
21	BOOL	1	Dock Mic Jack                           Off
22	BOOL	1	Mic Jack                                Off
23	BOOL	1	Internal Mic Phantom Jack               On
24	BOOL	1	Speaker Phantom Jack                    On
25	BOOL	1	Dock Headphone Jack                     Off
26	BOOL	1	Headphone Jack                          On
27	INT	4	Playback Channel Map                    3, 4, 0, 0 (range 0->36)
28	INT	2	Capture Channel Map                     0, 0 (range 0->36)
29	INT	2	PCM Playback Volume                     255, 255 (range 0->255)
yangyuqi@yangyuqi-T450:~/sound$ 
```

jsdosa, 2020-06-24:
@jerryyyq 
i think, you should ask to tinyalsa repo.
If you are testing on TizenRT, we will support the issue.

jerryyyq, 2020-06-24:
dear jsdosa, 
My Developer system is ubuntu, after my test OK, I will port to TizenRT, 
so, can you give me some advise？ thank you very much.

Taejun-Kwon, 2020-06-24:
> dear jsdosa,
> My Developer system is ubuntu, after my test OK, I will port to TizenRT,
> so, can you give me some advise？ thank you very much.

I think you tested pure tinyalsa on your ubuntu.
Tinyalsa in TizenRT is just brought signature of each function, to improve portability.
To use TizenRT, currently you need some specific device which i2s pin to connect dsp chip.
Will you develop embedded board?

jerryyyq, 2020-06-26:
Yes, I developer for  RK3288   RK 3399 and sdm845, these are ARM CPU, 32 or 64.

Taejun-Kwon, 2020-07-08:
It is very hard to help you , it is has depedency with dsp & platform.
So you can't hear your captured sound after capturing?
But I think first you'd better check your volume first.

jerryyyq, 2020-07-08:
I can hear the sound when I type: aplay -D 0 xxxx.wav.
now, I want captured the wav of playbacked on soundcard, 
so, I need some api to hook wav data from soundcard 0.
so, which api I can call them, example like: pcm_hook....... ?
thank you very much.