Samsung/TizenRT #4351: Compile failed when adding iotjs external lib for ESP32

weixiongmei, 2020-07-04:
Compile failed when adding iotjs external lib for ESP32, event trying to add the xtensa to the target-arch still failed. Please help~ Thanks
```py
    parser.add_argument('--target-arch',
        choices=['arm', 'xtensa', 'x86', 'i686', 'x86_64', 'x64'],
        default=platform.arch(),
        help='Specify the target architecture: '
             '%(choices)s (default: %(default)s)')
```

Errors
```shell
 --jerry_profile: /root/tizenrt/external/iotjs/deps/jerry/jerry-core/profiles/es5.1.profile
 --no_init_submodule: True
 --build_libs: /root/tizenrt/external/iotjs/build/xtensa-tizenrt/release/lib
 --iotjs_exclude_module: set([])
 --jerry_memstat: False
 --buildlib: True
 --clean: False
 --external_include_dir: ['/root/tizenrt/os/net/lwip/src/include']
 --external_static_lib: []
 --experimental: False
 --no_check_valgrind: False

==> Build IoT.js

    build_iotjs(options)
  File "/root/tizenrt/os/../external/iotjs/tools/build.py", line 423, in build_iotjs
    cmake_opt.extend(build_cmake_args(options))
  File "/root/tizenrt/os/../external/iotjs/tools/build.py", line 299, in build_cmake_args
    compile_flags += config_compile_flags['arch'][options.target_arch]
KeyError: 'xtensa'
make[1]: *** [build] Error 1
Makefile:35: recipe for target 'build' failed
make[1]: Leaving directory '/root/tizenrt/external/iotjs/config/tizenrt'
LibTargets.mk:214: recipe for target '../external/libexternal.a' failed
make: *** [../external/libexternal.a] Error 2
```

sunghan-chang, 2020-07-05:
@weixiongmei Thank you for the issue. Let me check.

sunghan-chang, 2020-07-06:
@weixiongmei Sorry but it needs to be supported by iotjs. That is difficult to be done by us.

sunghan-chang, 2020-07-06:
@haesik If possible, could you share the plan to support xtensa architecture with esp32 board on IoT.js project?

weixiongmei, 2020-07-06:
@sunghan-chang, I had have port all the ESP32 drivers for Jerryscript, but I couldn't get any success for the iotjs, because I'm not familiar with CMAKE, if anyone can help to get a successful compilation for ESP32, I can help to port all the drivers for it.

sunghan-chang, 2020-07-10:
@weixiongmei Sorry for late reply. Could you let us know your usages of  iotjs? We are working on micropython so that we could manage this. But I can't guarantee it on iotjs.