Samsung/TizenRT #4399: ARTIK055s programmer's manual

DevNaga, 2020-07-23:
Hello,

I am an owner of the ARTIK055s EVal board. I would require a programmer's manual. I have been looking over the google information relevant to Artik and it does seem that you have closed Artik business - however, i would like to have the programmer's manual in handy for reference purposes. Your Artik.io website does not function - so reaching out here for the help - could you let me know the best way to get the programmer's manual ?

Thanks,
Dev

sunghan-chang, 2020-07-24:
Hello @DevNaga .
Thank you for the issue. We are Samsung but TizenRT is managed in different business division.
ARTIK is just one of chip we support and we don't manage ARTIK itself. Let me leave a query to who managed (but he does not manage ARTIK now).
@JunhwanPark Hello. Is this possible to support?

DevNaga, 2020-07-24:
Hello @sunghan-chang  thank you. I really look forward to get a hold of the programmer's manual (aka. register definitions, i/o interfaces, module definitions etc and the schematics of the Artik 055s). I am still developing software for hobby and learning purposes - so please let me know the next steps.

sunghan-chang, 2020-07-24:
@DevNaga Let's wait his response first.

JunhwanPark, 2020-07-27:
Hi @DevNaga Nice to e-meet you. (@sunghan-chang long time no see~ )
artik.io has stopped, and now the artik document cannot be found. I will find out if I have it.

DevNaga, 2020-07-27:
Hello @JunhwanPark  Thank you nice to e-meet you as well. If found, could you please pass me at devendra(dot)aaru(at)gmail(dot)com ?

JunhwanPark, 2020-07-28:
@DevNaga Maybe @sunghan-chang will help you. I gave him the documents from artik.io.
Thx for your interest. :+1:

sunghan-chang, 2020-07-30:
Hello @DevNaga .
I tried to upload them into our repository but I can't do that with over 1MB size.
And because of company security, we can't send a mail to outside.
Let me find the way.

DevNaga, 2020-07-30:
Hello @sunghan-chang, 

i created a private repo here : https://github.com/DevNaga/ARTIK055s_manual_Samsung.

Added you here. You could upload file here. Hope this is right approach.

sunghan-chang, 2020-08-03:
@DevNaga There is a problem to upload some size of files. That's not our repository setting so that it's the same in yours. I googled and did what they suggested, but it did not work. Let me check it more.
FYI
```
error: RPC failed; HTTP 403 curl 22 The requested URL returned error: 403 Forbidden
fatal: the remote end hung up unexpectedly
```

sunghan-chang, 2020-08-10:
@DevNaga Sorry for much delay. Could you find them in https://github.com/Taejun-Kwon/TizenRT/tree/artik_doc/build/configs/artik055s?

DevNaga, 2020-08-11:
@sunghan-chang  i try to download the pdfs.. but all are damaged when i try to open them in Adobe viewer. the firefox viewer dont work as well.

madmax440, 2022-04-24:
@sunghan-chang thanks anyways.. i still do not have documentation even after 2 years. It does seem few units are still on sale in amazon etc. It would be great if Samsung does not kill the documentation this fast next time when they fail.