Samsung/TizenRT #4424: ESP32 Fails to boot: both esp32_DevKitC/tash_wlan and esp32_DevKitC/hello_with_tash configs

irsanvanwel, 2020-08-05:
Hi Everyone,

Awesome project! Really keen to get it running but I never get into `TASH`.

Please see the output on the Esp32 below:
```
��ets Jun  8 2016 00:22:57G early entropy source...

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5752
load:0x40078000,len:9020
load:0x40080000,len:6064
entry 0x40080330
I (29) boot: ESP-IDF v3.1-dirty 2nd stage bootloader
I (29) boot: compile time 18:12:49
I (29) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00280000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x17f3c ( 98108) map
I (126) esp_image: segment 1: paddr=0x00027f64 vaddr=0x3ffb0000 size=0x018e0 (  6368) load
I (129) esp_image: segment 2: paddr=0x0002984c vaddr=0x40080000 size=0x00400 (  1024) load
I (134) esp_image: segment 3: paddr=0x00029c54 vaddr=0x40080400 size=0x063bc ( 25532) load
I (153) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x7dd0c (515340) map
I (334) esp_image: segment 5: paddr=0x000add2c vaddr=0x400867bc size=0x02c6c ( 11372) load
I (339) esp_image: segment 6: paddr=0x000b09a0 vaddr=0x400c0000 size=0x00000 (     0) load
I (346) boot: Loaded app from partition at offset 0x10000
I (346) boot: Disabling RNG early entropy source...
xtensa_panic: Unhandled Exception 1073624984 task: hpwork
xtensa_dumpstate: sp:         3ffe3718
xtensa_dumpstate: stack base: 3ffe3b88
xtensa_dumpstate: stack size: 0000083c
xtensa_stackdump: 3ffe3700: 00000008 4584daff 800d607c 3ffe3758 0000003a 3f400ef3 fc1887fd a6b956d2
xtensa_stackdump: 3ffe3720: 00000008 c02cb7e0 7a4c4bab 12d2b1dd 00000008 c02cb7e0 7a4c4bab 12d2b1dd
xtensa_stackdump: 3ffe3740: 3f400f50 3ffe3798 400809a9 3ffe3778 3ffe3798 3ffe3798 3f400f50 3ffe3798
xtensa_stackdump: 3ffe3760: 3ffe2f14 3f400ef3 00040023 3ffe37b8 3ffe37b8 50000000 98da29aa 38b2be0f
xtensa_stackdump: 3ffe3780: b709521e 87205b3f e14f41a3 f7e9660f 9f6a02e8 ba40cd7f 800d6074 3ffe38e8
xtensa_stackdump: 3ffe37a0: 00000003 3f400ef3 00040023 3ffe3978 3ffe3978 40000000 3ffe3b18 00000000
xtensa_stackdump: 3ffe37c0: 800d6074 3ffe38e8 00000003 3f400ef3 3f400f50 3ffe3958 3ffe2f14 00000007
xtensa_stackdump: 3ffe37e0: 00000005 0000001d 84f4e92f 00000000 00000000 00000000 4008098c 3ffe3798
xtensa_stackdump: 3ffe3800: 00000004 0000001d 3ffe3918 3ffe38f8 00000020 00000005 4008098c 3ffe37b8
xtensa_stackdump: 3ffe3820: 4518be9a 4d1b1a6a 80104dcc 3ffe3898 00000003 3f400ef3 f6ff6012 159f2d4b
xtensa_stackdump: 3ffe3840: 00f5b6ca 8311c175 3ffe3918 3ffe38f8 00000008 3ffe38f8 3ffe3918 3f400ef5
xtensa_stackdump: 3ffe3860: 3f400f50 00000001 3ffe3898 c02adbc7 7fe3138d 3afd8544 3da5192f 8a4190f5
xtensa_stackdump: 3ffe3880: e2f9e11d 8b567416 800d6074 3ffe38e8 00000003 3f400ef3 40105b04 4014cb84
xtensa_stackdump: 3ffe38a0: 00000000 3ffe3918 3ffe38f8 00000008 ec415e3a f5d40e54 3ffe3918 3ffe38f8
xtensa_stackdump: 3ffe38c0: 00000008 f9aa9ced 80104da6 3ffe3838 00000000 3f400f50 0000000c 3f400f5c
xtensa_stackdump: 3ffe38e0: 0000000c 00000000 3ffe3918 3ffe38f8 00000008 89bebff1 b2075766 9aa923de
xtensa_stackdump: 3ffe3900: 3f400f50 3ffe3958 3ffe2f14 00000007 5dafbc09 a07ed59d 3f400f50 3ffe3958
xtensa_stackdump: 3ffe3920: 3ffe2f14 00000007 96d6828e 84f4e94f e37b8cab e9c9623f 043d6147 2bb85367
xtensa_stackdump: 3ffe3940: 30c4ae64 9a555142 400d5e01 4b7c8279 b70545ac d928f061 00040023 3ffe3798
xtensa_stackdump: 3ffe3960: 3ffe3798 00000006 800d3674 3ffe3b48 00000001 3ffe3b8c 00000000 3ffb5e68
xtensa_stackdump: 3ffe3980: 80101800 3ffe3ac8 3ffe3b18 00000000 00000000 3ffe2e10 00060023 00020000
xtensa_stackdump: 3ffe39a0: 00000002 0000001d 00060000 00000000 00000000 00000000 4008098c 3ffe3958
xtensa_stackdump: 3ffe39c0: 00000002 0000001c 00060014 00000000 00000000 00000000 4008098c 3ffe3978
xtensa_stackdump: 3ffe39e0: 00000000 00000000 00000000 00000000 800d3674 3ffe3b48 00000001 3ffe3b8c
xtensa_stackdump: 3ffe3a00: 00000000 fffffff0 00000000 3ffb5e68 00000001 0000001d fffffff0 00000000
xtensa_stackdump: 3ffe3a20: 00000000 00000000 4008098c 3ffe39c8 00000001 0000001d 880bceae 00000000
xtensa_stackdump: 3ffe3a40: 00000000 00000000 4008098c 3ffe39e8 65b622ac 4d15ce52 9d0035d6 3343139b
xtensa_stackdump: 3ffe3a60: 97ab8c8a 35bfe257 1a87a0f0 3c8605bb 470a7749 5772067d 00000000 3ffe2e10
xtensa_stackdump: 3ffe3a80: 00060023 00020000 95962ab5 ba0dcb72 53cf6de1 1673b74b 00000000 6195bd70
xtensa_stackdump: 3ffe3aa0: a361532f 3e55cc4f 956da95f 44b00050 1f995977 4efe61c1 d91ef7a6 cf2dcbf2
xtensa_stackdump: 3ffe3ac0: b6e7554e 86b65e4c 00000000 643cf192 d3d0f877 6bb4a2b7 ee8805e0 85055bcc
xtensa_stackdump: 3ffe3ae0: acf189da a17319f9 800d5bbb 3ffe3b18 3ffb6158 3ffb6158 a4dba621 3fdaf57c
xtensa_stackdump: 3ffe3b00: 738f3eb4 bb925597 4fa45242 9af2f450 ec6cb411 2664403d 00020000 b138587f
xtensa_stackdump: 3ffe3b20: fc8e8fc9 ee8e5f1d 400d5e01 00000109 00000022 00000004 400d5e01 3ffe3b88
xtensa_stackdump: 3ffe3b40: 00005554 0000000d 71ecbd4d b8627a45 30303030 34303063 9fdc5994 880bcece
xtensa_stackdump: 3ffe3b60: bd8215a2 bdd66b5f 400d5e01 09000001 00000020 00000004 800d3629 3ffde6b0
xtensa_stackdump: 3ffe3b80: 00000090 3ffe2e10 e17c4d76 3ffe3b94 00000000 6f777068 28006b72 128e32fb
xtensa_registerdump:    PC: 800d6074    PS: 3ffe38e8
xtensa_registerdump:    A0: 00000003    A1: 3f400ef3    A2: 00040023    A3: 3ffe3978
xtensa_registerdump:    A4: 3ffe3978    A5: 40000000    A6: 3ffe3b18    A7: 00000000
xtensa_registerdump:    A8: 800d6074    A9: 3ffe38e8   A10: 00000003   A11: 3f400ef3
xtensa_registerdump:   A12: 3f400f50   A13: 3ffe3958   A14: 3ffe2f14   A15: 00000007
xtensa_registerdump:   SAR: 00000005 CAUSE: 0000001d VADDR: 84f4e92f
xtensa_registerdump:  LBEG: 00000000  LEND: 00000000  LCNT: 00000000
xtensa_registerdump:  TMP0: 4008098c  TMP1: 3ffe3798
```
The `hello_with_tash` example doesn't boot either, never get to see the `TASH` console. 

Am I missing something?

irsanvanwel, 2020-08-06:
@jeongarmy @jeongchanKim Just some extra information: this is only happening on `Master` and `v3.1_PRE`. `3.0_GBM` is working fine, so this seems to be a regression caused by changes since then. I am going to compare the differences somewhere the coming days. Let me know if there is any particular area I should focus on. Also feel free to let me know if there is anything I can assist with.

Kashyapkoshti, 2020-08-28:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32