Samsung/TizenRT #4624: Does tizenrt support fpclassify ?

HarishSRIB, 2020-11-24:
I am porting a library that is developed on Linux platform to TizenRT platform. I am getting an undefined linker error to fpclassify method. Does any body know what is the problem here ? Does it need any configuration variable to enabled ? 

BTW  I am using gcc-arm-none-eabi-9-2019-q4-major tool chain to build.

jeongchanKim, 2020-11-24:
Hello @HarishSRIB . Which config do you use now?
And as my checking, `fpclassify` is in cpp. Did you enable cpp also?

HarishSRIB, 2020-11-24:
@jeongchanKim, could you please elaborate more on enabling cpp. Is there a dedicated config variable to enable cpp in the config file?

sijupunnnoose, 2020-11-24:
@HarishSRIB please discuss with @sangam-swami for more details. He would help you with CPP enablement.  Please share which board/config you are trying to build for.

sunghan-chang, 2020-11-30:
@HarishSRIB @sangam-swami Is there any update on this?

HarishSRIB, 2020-12-04:
@sunghan-chang, I got busy with some other project deadlines.

sangam-swami, 2020-12-09:
The current fpclassify issue in MTK board is because cpp is not ported to MTK board.
@HarishSRIB you can use below options.
1. I checked by disabling the fpclassify call in your code base, it got compiled and created tinyara.bin.

closed 2021-04-12 by sunghan-chang