Samsung/TizenRT #5602: Unusable source tree

madmax440, 2022-08-15:
Hi,

I am facing a lot of difficulties in compiling firmware image for Artik055s.

So far i am following the manual setup. Below are my build steps.

```bash
1. cd TizenRT/os/tools
2. ./configure.sh artik055s/st_things
3. cd ..
4. make menuconfig # do nothing.. just exit and save config
5. make
```

After step 5, build fails at various places in master. I have tried to resolve one by one but there seem too many build failures.

Can someone help?

Thanks

sunghan-chang, 2022-08-16:
@madmax440 Hello. Sorry for inconvenience.
@jeongarmy Could you check it?

jeongarmy, 2022-08-16:
@sangwon03 As I checked, many build errors about network configurations happens. Could you please check this?

madmax440, 2022-08-27:
Hello,

With master at  bca7d1a811b8fe303da53e7492c25a1b4e9505c8 i still have the below build error.

```bash
CC:  /home/devnaga/work/TizenRT/os/arch/boardctl.c
AR:   artik05x_boot.o artik05x_tash.o artik05x_pwm.o artik05x_progmem.o artik05x_wlan.o partitions.o boardctl.o
make[2]: Leaving directory '/home/devnaga/work/TizenRT/os/board/artik05x/src'
LD: tinyara
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libnet.a(netdev_wifi.o): in function `_trwifi_event_handler':
/home/devnaga/work/TizenRT/os/net/netmgr/netdev_wifi.c:307: undefined reference to `lwnl_postmsg'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_info.o): in function `wifimgr_get_info':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_info.c:58: undefined reference to `wifi_utils_get_info'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_wifimgr_scan':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:303: undefined reference to `wifi_utils_scan_ap'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:318: undefined reference to `wifi_utils_scan_ap'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_wifimgr_deinit':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:195: undefined reference to `wifi_utils_deinit'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_wifimgr_disconnect_ap':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:255: undefined reference to `wifi_utils_disconnect_ap'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_wifimgr_run_softap':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:277: undefined reference to `wifi_utils_start_softap'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_wifimgr_connect_ap':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:236: undefined reference to `wifi_utils_connect_ap'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_wifimgr_stop_softap':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:293: undefined reference to `wifi_utils_stop_softap'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_wifimgr_run_sta':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:203: undefined reference to `wifi_utils_start_sta'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:207: undefined reference to `wifi_utils_set_autoconnect'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_handler_on_uninitialized_state':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:331: undefined reference to `wifi_utils_init'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:345: undefined reference to `wifi_utils_set_autoconnect'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_handler_get_stats':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:560: undefined reference to `wifi_utils_ioctl'
arm-none-eabi-ld: /home/devnaga/work/TizenRT/os/../build/output/libraries/libframework.a(wifi_manager_state.o): in function `_handler_set_powermode':
/home/devnaga/work/TizenRT/framework/src/wifi_manager/wifi_manager_state.c:593: undefined reference to `wifi_utils_ioctl'
make[1]: *** [Makefile:200: ../build/output/bin/tinyara] Error 1
make[1]: Leaving directory '/home/devnaga/work/TizenRT/os/arch/arm/src'
make: *** [Makefile.unix:460: pass2] Error 2

```