Samsung/TizenRT #5778: Security: task_terminate_unloaded function lack of sched_lock() will lead to race condition

peanuts62, 2023-03-13:
when calling this function, it will into kernel state, if it's time have another thread will lead to privilege escalation.

```
int task_terminate_unloaded(FAR struct tcb_s *tcb)
{
	irqstate_t saved_state;

#if defined(CONFIG_APP_BINARY_SEPARATION) && defined(CONFIG_ARM_MPU)
	/* Disable mpu regions when the binary is unloaded if its own mpu registers are set in mpu h/w. */
	if (IS_BINARY_MAINTASK(tcb) && up_mpu_check_active(&tcb->mpu_regs[0])) {
#ifdef CONFIG_OPTIMIZE_APP_RELOAD_TIME
		for (int i = 0; i < MPU_REG_NUMBER * MPU_NUM_REGIONS; i += MPU_REG_NUMBER) {
			up_mpu_disable_region(&tcb->mpu_regs[i]);
		}
#else
		up_mpu_disable_region(&tcb->mpu_regs[0]);
#endif
	}
#endif

#if defined(CONFIG_SCHED_ATEXIT) || defined(CONFIG_SCHED_ONEXIT)
	task_atexit(tcb);
	task_onexit(tcb, EXIT_SUCCESS);
#endif
#ifdef HAVE_TASK_GROUP
	group_leave(tcb);
#endif
	sig_cleanup(tcb);

	saved_state = irqsave();
	dq_rem((FAR dq_entry_t *)tcb, (dq_queue_t *)g_tasklisttable[tcb->task_state].list);
	irqrestore(saved_state);

#ifdef CONFIG_TASK_MONITOR
	/* Unregister this pid from task monitor */
	task_monitor_unregester_list(tcb->pid);
#endif
#ifdef CONFIG_PREFERENCE
	preference_clear_callbacks(tcb->pid);
#endif
	tcb->flags |= TCB_FLAG_EXIT_PROCESSING;

	/* No need to release the unloading thread's stack,
	 * because whole heap memory will be released at one go.
	 * So marking alloc pointer to NULL for skipping to release.
	 */
	tcb->stack_alloc_ptr = NULL;

	/* Deallocate its TCB */

	return sched_releasetcb(tcb, tcb->flags & TCB_FLAG_TTYPE_MASK);
}

```

Should do it like other function:
```
int task_terminate(pid_t pid, bool nonblocking)
{
	FAR struct tcb_s *dtcb;
	irqstate_t saved_state;
	trace_begin(TTRACE_TAG_TASK, "task_terminate");

	/* Make sure the task does not become ready-to-run while we are futzing with
	 * its TCB by locking ourselves as the executing task.
	 */

	sched_lock();<------------------------------------lock it

	/* Find for the TCB associated with matching PID */

	dtcb = sched_gettcb(pid);
	if (!dtcb) {
		/* This PID does not correspond to any known task */

		sched_unlock();
		trace_end(TTRACE_TAG_TASK);
		return -ESRCH;
	}

	/* Verify our internal sanity */

	if (dtcb->task_state == TSTATE_TASK_RUNNING || dtcb->task_state >= NUM_TASK_STATES) {
		sched_unlock();
		PANIC();
	}

#if defined(CONFIG_APP_BINARY_SEPARATION) && defined(CONFIG_ARM_MPU)
	/* Disable mpu regions when the binary is unloaded if its own mpu registers are set in mpu h/w. */
	if (IS_BINARY_MAINTASK(dtcb) && up_mpu_check_active(&dtcb->mpu_regs[0])) {
#ifdef CONFIG_OPTIMIZE_APP_RELOAD_TIME
		for (int i = 0; i < MPU_REG_NUMBER * MPU_NUM_REGIONS; i += MPU_REG_NUMBER) {
			up_mpu_disable_region(&dtcb->mpu_regs[i]);
		}
#else
		up_mpu_disable_region(&dtcb->mpu_regs[0]);
#endif
	}
#endif

	/* Perform common task termination logic (flushing streams, calling
	 * functions registered by at_exit/on_exit, etc.).  We need to do
	 * this as early as possible so that higher level clean-up logic
	 * can run in a healthy tasking environment.
	 *
	 * In the case where the task exits via exit(), task_exithook()
	 * may be called twice.
	 *
	 * I suppose EXIT_SUCCESS is an appropriate return value???
	 */

	task_exithook(dtcb, EXIT_SUCCESS, nonblocking);

	/* Remove the task from the OS's tasks lists. */

	saved_state = irqsave();
	dq_rem((FAR dq_entry_t *)dtcb, (dq_queue_t *)g_tasklisttable[dtcb->task_state].list);
	dtcb->task_state = TSTATE_TASK_INVALID;
#ifdef CONFIG_TASK_MONITOR
	/* Unregister this pid from task monitor */
	task_monitor_unregester_list(pid);
#endif
#ifdef CONFIG_PREFERENCE
	preference_clear_callbacks(pid);
#endif
	irqrestore(saved_state);

	/* At this point, the TCB should no longer be accessible to the system */

	sched_unlock();

	trace_end(TTRACE_TAG_TASK);

	/* Deallocate its TCB */

	return sched_releasetcb(dtcb, dtcb->flags & TCB_FLAG_TTYPE_MASK);
}
```

sunghan-chang, 2023-03-14:
@peanuts62 Thank you for the issue. Let me check it.
@kishore-sn Could you check this with Prabu?

sunghan-chang, 2023-03-20:
@kishore-sn Any update on this?

sunghan-chang, 2023-03-31:
@peanuts62 This was merged by PR #5793 . Thank you for the issue rasie.

closed 2023-03-31 by sunghan-chang

peanuts62, 2023-03-31:
Can you request CVE ID for the vulnerability?

peanuts62, 2023-03-31:
@sunghan-chang