Samsung/TizenRT #5800: there is no NULL check in lwip sockets

hasw7569, 2023-04-05:
Hello, TizenRT!

I found a problem while doing tcp communication with TizenRT.

That is, the `trycheck_selwait_socket()` function does not have a NULL check.

`os/net/lwip/src/api/sockets.c`

```
struct lwip_sock *trycheck_selwait_socket(int s, pid_t pid)
{
	struct lwip_sock *sock;
	sock = tryget_socket_by_pid(s, pid);

	if (!sock->conn) {
		return sock;
	}
	return NULL;
}
```

if `struct lwip_sock *sock` has NULL value, then it makes error.

I think we need a NULL check code. Could you update the code if I'm right?

Taejun-Kwon, 2023-04-05:
I agree with you Let's contribute by yourself!

hasw7569, 2023-04-07:
plz check my PR : )
I just added simple NULL check
[https://github.com/Samsung/TizenRT/pull/5804](https://github.com/Samsung/TizenRT/pull/5804)

hasw7569, 2023-04-24:
solved