Samsung/TizenRT #5845: Compile error on stm32: '_sint_heap_start' undeclared

szsam, 2023-06-09:
```
make[1]: Entering directory '/home/mjshen/IVOS/repos/TizenRT/os/arch/arm/src'
CC:  chip/stm32l4_start.c
chip/stm32l4_start.c:81:33: error: '_sint_heap_start' undeclared here (not in a function); did you mean '__sint_heap_start'? 
81 | #define HEAP_BASE  ((uintptr_t)&_sint_heap_start)
   |                                 ^~~~~~~~~~~~~~~~
chip/stm32l4_start.c:92:35: note: in expansion of macro 'HEAP_BASE'
92 | const uintptr_t g_idle_topstack = HEAP_BASE;
   |                                   ^~~~~~~~~  
```

### Reproduction step
``` bash
./tools/configure.sh   stm32l4r9ai-disco/pm
make
```

### Environment
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
Ubuntu 20.04

sunghan-chang, 2023-06-12:
@r-prabu could you resolve this?

sunghan-chang, 2023-07-28:
@szsam Really sorry for late. This is resolved by https://github.com/Samsung/TizenRT/pull/5874. If you meet any other problem, please reopen the issue.

closed 2023-07-28 by sunghan-chang