Samsung/TizenRT #5910: RTL8721CMS link file questtion?

Jan9401, 2023-08-25:
I used RTL8721csm board to test TizenRT rtos, I enter a question. list below

#[MODULE_BOOT-LEVEL_INFO]:IMG1 ENTER ROMSUB:2
bootloader_version km0_bootloader_ver_07729b9_2022/01/18-14:40:05
#utoboot in 50 milliseconds
Normal boot
BP1 data valid, version:1
BP2 version invalid
BP1 CRC32 match, attached CRC: 0x9730a7bd, calculated CRC: 0x9730a7bd
BP1 ota2 address invalid
Both BP invalid
Recovery start
[MODULE_BOOT-LEVEL_INFO]:OTA1 USE, version: 200204
[MODULE_BOOT-LEVEL_INFO]:IMG2 DATA[0xc010cc0:5840:0x85000]
[MODULE_BOOT-LEVEL_INFO]:IMG2 SIGN[RTKWin(8500c)]
[MODULE_BOOT-LEVEL_INFO]:IMG2 ENTRY[0x85000:0xc00d63d]
[MODULE_BOOT-LEVEL_INFO]:KM0 BOOT_IMG2 BOOT REASON: 0 
bootloader_version km0_application_ver_963c433_2023/04/06-16:40:45
calibration_ok:[2:19:11] 
bootloader_version km4_bootloader_ver_1275267_2021/12/15-21:10:03
[MODULE_BOOT-LEVEL_INFO]:IMG1 ENTER MSP:[1007fffc]
IMG2 SBOOT OFF
[MODULE_BOOT-LEVEL_INFO]:IMG3 RAM_S:[0x10077000:5632:0x8190020]
[MODULE_BOOT-LEVEL_INFO]:IMG3 NSC:[0x1007b000:480:0x8191640]
[MODULE_BOOT-LEVEL_INFO]:IMG2 DATA[0xe0d3e84:155920:0x10005000]
[MODULE_BOOT-LEVEL_INFO]:IMG2 SIGN[RTKWin(1000500c)]
[MODULE_BOOT-LEVEL_INFO]:IMG2 ENTRY[0x10005000:0xe000c9d]
[MODULE_BOOT-LEVEL_INFO]:Start NonSecure @ 0xe000c9c ...
BOOT_IMG3: BSS [10078600~10079e2c] SEC: 1 
[MODULE_BOOT-LEVEL_INFO]:IMG2 PSRAM_NS:[0xe0f9fb4:988:0x2000000]
[MODULE_BOOT-LEVEL_INFO]:IMG3 PSRAM_S:[0x23c0000:129792:0x8191840]
REG_HS_RFAFE_IND_VIO1833 (0 is 1.8V): 1
mpu_showtype: Unified MPU Regions: data=8 instr=0
up_allocate_kheap: start = 0x1005ba00 size = 112128
up_add_kregion: start = 0x2001f1c size = 3924196
Manufacturer : 28 memory type : 112 capacity : 21
up_busfault: #########################################################################
up_busfault: PANIC!!! Bus fault at instruction: 0x0e0013a6
up_busfault: FAULT TYPE: IMPRECISERR (Imprecise data access error occurred).
up_busfault: FAULT ADDRESS: Unable to determine fault address.
up_busfault: FAULT REGS: CFAULTS: 0x00000400 BFAR: 0x00000000
up_busfault: #########################################################################
up_assert: Assertion failed at file:armv8-m/up_busfault.c line: 126 task: Idle Task
up_dumpstate: Code asserted in normal thread!
up_dumpstate: Current SP is User Thread SP: 10004e48
up_dumpstate: User stack:
up_dumpstate:   base: 10004ff8
up_dumpstate:   size: 00000400
up_stackdump: 10004e40: xxxxxxxx xxxxxxxx 48006000 00497000 000007fe 00497000 00000504 10005591
up_stackdump: 10004e60: 00000000 0e02bdcd 00000497 0e08636d 1005fde0 00004020 00000603 0e020b3d
up_stackdump: 10004e80: 1005fde0 00000603 1005fe40 00000000 00000000 0e09a09b 10004ebc 10004ebc
up_stackdump: 10004ea0: 00000000 0e020951 00000001 00000000 10004ed4 00000000 00000000 1002b7f4
up_stackdump: 10004ec0: 00000000 0e0d3e64 0e000000 0e09a699 00000000 100630b0 03fb0400 0f330ff0
up_stackdump: 10004ee0: 0e002000 0000000d 10004f08 0000000e 1002b7f4 00000000 00000000 0e02c481
up_stackdump: 10004f00: 00000007 10004ef1 7665642f 616d732f 70307472 00000037 1002b7f4 0e0279cf
up_stackdump: 10004f20: 00000007 00000000 00000000 00000000 00000000 0e027a23 1002c110 10012809
up_stackdump: 10004f40: 1002b710 00000000 1002b9b8 00000000 0e0d3e64 100126a9 10004fa8 00000020
up_stackdump: 10004f60: 1002b710 00000000 1002b9b8 00000000 0e0d3e64 0e000000 00000000 00000000
up_stackdump: 10004f80: ffffffb8 00000000 00000002 1002b79c 1005ed2c 00000000 100127b1 0e002833
up_stackdump: 10004fa0: 100127ba 01000000 1002befc 00000002 1002b7f4 0e005235 00000000 1002b79c
up_stackdump: 10004fc0: 1005debc 00000000 00000000 0e001495 00000003 1002b710 1002b710 10010ed7
up_stackdump: 10004fe0: 1005ba00 0001b600 48012000 0e000dd9 02000000 1000500c feffffff 97d24d50
up_registerdump: R0: 00000001 00000000 ffffffff adbeefde 00000002 00000000 00497000 ffffffff
up_registerdump: R8: 0e0d3e64 0e000000 00000000 00000000 1010c95d 10004e48 0e001397 0e0013a6
up_registerdump: xPSR: 21000000 BASEPRI: 00000000 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffb8
task_show_alivetask_list: *******************************************
task_show_alivetask_list: List of all tasks in the system:
task_show_alivetask_list: *******************************************
task_show_alivetask_list:                            NAME |   PID |  PRI | TOTAL STACK | STACK ALLOC ADDR | TCB ADDR
task_show_alivetask_list: ---------------------------------------------------------------------------------------------------
task_taskdump:                       Idle Task |     0 |    0 |    1024 |       0x10004bfc | 0x1002b710
task_taskdump:                          hpwork |     1 |  224 |    2028 |       0x1005e490 | 0x1005de30
task_taskdump:                          lpwork |     2 |   50 |    2028 |       0x1005f300 | 0x1005eca0
mpu_show_regioninfo: ********************************************************************************************
mpu_show_regioninfo: REGION_NO.         BASE_ADDRESS    SIZE            STATUS          ACCESS (P/U)    EXECUTE     
mpu_show_regioninfo: ********************************************************************************************
mpu_show_regioninfo: 0                  0x10046280      0x157A0         ENABLED         RW/RW           X           
mpu_show_regioninfo: 1                  0x2000000       0x400020        ENABLED         RW/RW           X           
mpu_show_regioninfo: 2                  0x0             0x20            DISABLED        RW/NO           X           
mpu_show_regioninfo: 3                  0x0             0x20            DISABLED        RW/NO           X           
mpu_show_regioninfo: 4                  0x0             0x20            DISABLED        RW/NO           X           
mpu_show_regioninfo: 5                  0x0             0x20            DISABLED        RW/NO           X           
mpu_show_regioninfo: 6                  0x0             0x20            DISABLED        RW/NO           X           
mpu_show_regioninfo: 7                  0x0             0x20            DISABLED        RW/NO           X           
mpu_show_regioninfo: ********************************************************************************************
up_assert: Checking kernel heap for corruption...
mm_check_heap_corruption: Heap start = 0x1005ba00 end = 0x10076ff8
mm_check_heap_corruption: Heap start = 0x02001f20 end = 0x023bfff8
up_assert: No kernel heap corruption detected
up_assert: Assert location (PC) : 0x0e0013a6
up_assert: *******************************************
up_assert: Asserted TCB Info
up_assert: *******************************************
task_show_tcbinfo: State       : 3
task_show_tcbinfo: Flags       : 2
task_show_tcbinfo: Lock count  : 0
task_show_tcbinfo: Timeslice   : 0
task_show_tcbinfo: Waitdog     : 0
task_show_tcbinfo: WaitSem     : 0
task_show_tcbinfo: MsgwaitQ    : 0
task_show_tcbinfo: Sigdeliver  : 0

From rtl872x0csm spec, there is no 0x0E000000 address, but the ld script set KM4 IMAGE address 0x0E000000, I don't  understand that why can use this address.

sunghan-chang, 2023-08-28:
https://github.com/Samsung/TizenRT/blob/master/build/configs/rtl8721csm/scripts/rlx8721d_img2_ns.ld#L83

TizenRT is loaded in KM4 region. It has 0xE........ address.

Jan9401, 2023-08-28:
> https://github.com/Samsung/TizenRT/blob/master/build/configs/rtl8721csm/scripts/rlx8721d_img2_ns.ld#L83
> 
> TizenRT is loaded in KM4 region. It has 0xE........ address.

Yes, I see,the problem is that it will trigger a bus fault, PC is 0x0e0013a6, I further debug and found that a bus fault t occurred during IPC sendrequest,  I used to command "make download ALL" to download image. I don't know how to compile KM0.Is it related to the lack of KM0 image. Thanks for your reply

sunghan-chang, 2023-08-28:
> I don't know how to compile KM0.

We use prebuilt KM0, not to dynamically compile. 

> Is it related to the lack of KM0 image.

I don't think so. Your codes cause the issue. There is no issue on current version with RTL8721CSM.
We usually meet a busfault with null dereference.

Jan9401, 2023-08-28:
> > I don't know how to compile KM0.
> 
> We use prebuilt KM0, not to dynamically compile.
> 
> > Is it related to the lack of KM0 image.
> 
> I don't think so. Your codes cause the issue. There is no issue on current version with RTL8721CSM. We usually meet a busfault with null dereference.

Which defconfig are you using? I used cxxtest. My test steps below list:

./tools/configure.sh rtl8721csm/cxxtest
make 
make download ALL port=ttyUSB0

I can use the defconfig you tested to verify if there are any issues.

sunghan-chang, 2023-08-28:
> Which defconfig are you using? I used cxxtest. My test steps below list:
>
> ./tools/configure.sh rtl8721csm/cxxtest
> make
> make download ALL port=ttyUSB0
>
> I can use the defconfig you tested to verify if there are any issues.

Ok.
@prabu-r Could you check `rtl8721csm/cxxtest`?

abhishek-samsung, 2023-08-29:
Hello @Jan9401 , we have verified it on our side with the same config (```rtl8721csm/cxxtest```) on the latest commit (https://github.com/Samsung/TizenRT/commit/846ea7198d0c79bbd5a1785a8ab1c0a893f7f54b) and there is no issue. 

```
[MODULE_BOOT-LEVEL_INFO]:Start NonSecure @ 0xe000d0c ...
BOOT_IMG3: BSS [10078600~10079e2c] SEC: 1
[MODULE_BOOT-LEVEL_INFO]:IMG2 PSRAM_NS:[0xe0fb9b8:988:0x2000000]
[MODULE_BOOT-LEVEL_INFO]:IMG3 PSRAM_S:[0x23c0000:129792:0x8192840]
REG_HS_RFAFE_IND_VIO1833 (0 is 1.8V): 1
mpu_showtype: Unified MPU Regions: data=8 instr=0
up_allocate_kheap: start = 0x1005c0e0 size = 110368
up_add_kregion: start = 0x2001f1c size = 3924196
Manufacturer : 161 memory type : 64 capacity : 23
/dev/smart0p7 is mounted successfully @ /mnt
se_ameba_hal_init: [AMEBA_WRAPPER]ameba/security_ameba_wrapper_tz.c:59
The key slot is empty

Failed to decrypt 1st secure storage key

System Information:
        Version:
                Platform: 3.1   Binary: 200204
        Commit Hash: 846ea71
        Build User: root@=
        Build Time: 2023-08-29 09:37:34 IST
        System Time: 06 Dec 2011, 00:00:00 [s] UTC
TASH>>Hello, World!!

TASH>>cxxtest
TASH>>
============ Test ofstream ==============================
```

abhishek-samsung, 2023-08-29:
@Jan9401, could you please share the toolchain version being used? Could you also please share the details of the board being used? We also support build using docker (https://hub.docker.com/r/tizenrt/tizenrt) and it can be triggered using ```./dbuild.sh menu```. If possible could you please try the build with docker?

Jan9401, 2023-08-29:
> @Jan9401, could you please share the toolchain version being used? Could you also please share the details of the board being used? We also support build using docker [...]

My gcc version :gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10).
I used to RTL8720csm board, In theory, there should be no other difference between 8720 and 8721 except for the pins, But I saw log, I found that there were still differences in FLASh( Manufacturer : 161 memory type : 64 capacity : 23) . can you share your toolchain and complete testing steps?

sunghan-chang, 2023-08-29:
@Jan9401 We use ARM GCC 6.3.1, but we recommend to use docker instead of using ARM GCC directly. (TizenRT docker image contains it) Testing steps are just building, downloading and booting. That's it. After installing the docker, you can use it through `./dbuild.sh menu` at os folder. Please refer https://github.com/samsung/tizenrt#environment-setup.

Jan9401, 2023-08-29:
> @Jan9401 We use ARM GCC 6.3.1, but we recommend to use docker instead of using ARM GCC directly. (TizenRT docker image contains it) Testing steps are just building, downloading and booting. That's [...]

I download ARM GCC 6.31, I foud that there is no issue.I am not familiar with gcc, so I am not sure how this bus fault was generated.

Jan9401, 2023-08-29:
I encountered another issue, I entered reboot and prompted 'All OTA addresses not valid, recovery failed'.

sunghan-chang, 2023-08-29:
@Jan9401 It's impossible to give any guidance with this information. Could you leave your logs?

sunghan-chang, 2023-08-29:
> I download ARM GCC 6.31, I foud that there is no issue.I am not familiar with gcc, so I am not sure how this bus fault was generated.

Because of that, we provide a script to use the TizenRT docker. When you use it, you don't need to install any tool including ARM GCC. When we update a version of installed tool or request a new tool installation, you can do that with docker version change (And docker version change will be done by our script.).

Jan9401, 2023-08-29:
> @Jan9401 It's impossible to give any guidance with this information. Could you leave your logs?

sorry for less information. I get complete log show below:

#[MODULE_BOOT-LEVEL_INFO]:IMG1 ENTER ROMSUB:2
bootloader_version km0_bootloader_ver_07729b9_2022/01/18-14:40:05
#utoboot in 50 milliseconds
Normal boot
BP1 data valid, version:1
BP2 version invalid
BP1 CRC32 match, attached CRC: 0x9730a7bd, calculated CRC: 0x9730a7bd
BP1 ota2 address invalid
Both BP invalid
Recovery start
[MODULE_BOOT-LEVEL_INFO]:OTA1 USE, version: 200204
[MODULE_BOOT-LEVEL_INFO]:IMG2 DATA[0xc010cc0:5840:0x85000]
[MODULE_BOOT-LEVEL_INFO]:IMG2 SIGN[RTKWin(8500c)]
[MODULE_BOOT-LEVEL_INFO]:IMG2 ENTRY[0x85000:0xc00d63d]
[MODULE_BOOT-LEVEL_INFO]:KM0 BOOT_IMG2 BOOT REASON: 0 
bootloader_version km0_application_ver_963c433_2023/04/06-16:40:45
calibration_ok:[2:19:11] 
bootloader_version km4_bootloader_ver_1275267_2021/12/15-21:10:03
[MODULE_BOOT-LEVEL_INFO]:IMG1 ENTER MSP:[1007fffc]
IMG2 SBOOT OFF
[MODULE_BOOT-LEVEL_INFO]:IMG3 RAM_S:[0x10077000:5632:0x8191020]
[MODULE_BOOT-LEVEL_INFO]:IMG3 NSC:[0x1007b000:480:0x8192640]
[MODULE_BOOT-LEVEL_INFO]:IMG2 DATA[0xe0d4f38:157760:0x10005000]
[MODULE_BOOT-LEVEL_INFO]:IMG2 SIGN[RTKWin(1000500c)]
[MODULE_BOOT-LEVEL_INFO]:IMG2 ENTRY[0x10005000:0xe000cd9]
[MODULE_BOOT-LEVEL_INFO]:Start NonSecure @ 0xe000cd8 ...
BOOT_IMG3: BSS [10078600~10079e2c] SEC: 1 
[MODULE_BOOT-LEVEL_INFO]:IMG2 PSRAM_NS:[0xe0fb798:988:0x2000000]
[MODULE_BOOT-LEVEL_INFO]:IMG3 PSRAM_S:[0x23c0000:129792:0x8192840]
REG_HS_RFAFE_IND_VIO1833 (0 is 1.8V): 1
mpu_showtype: Unified MPU Regions: data=8 instr=0
up_allocate_kheap: start = 0x1005c140 size = 110272
up_add_kregion: start = 0x2001f1c size = 3924196
Manufacturer : 28 memory type : 112 capacity : 21
smartfs_partno: 7, romfs_partno: 0
se_ameba_hal_init: [AMEBA_WRAPPER]ameba/security_ameba_wrapper_tz.c:59
The key slot is empty

Failed to decrypt 1st secure storage key

System Information:
        Version:
                Platform: 3.1   Binary: 200204
        Commit Hash: 5e87c59
        Build User: root@xiaoming-Mi-Laptop-Pro-15
        Build Time: 2023-08-29 15:07:04 CST
        System Time: 06 Dec 2011, 00:00:00 [s] UTC 
TASH>>Hello, World!!

TASH>>
TASH>>reboot 
boardctl: Board will Reboot now.
#[MODULE_BOOT-LEVEL_INFO]:IMG1 ENTER ROMSUB:0
bootloader_version km0_bootloader_ver_07729b9_2022/01/18-14:40:05
#utoboot in 50 milliseconds
Normal boot
BP1 data valid, version:1
BP2 version invalid
BP1 CRC32 match, attached CRC: 0x9730a7bd, calculated CRC: 0x9730a7bd
BP1 ota2 address invalid
Both BP invalid
Recovery start
All OTA addresses not valid, recovery fail

I reboot system, it tips OTA address not valid .I do not see any information about OTA.

Jan9401, 2023-08-31:
I don't have this log.  "dev/smart0p7 is mounted successfully @ /mnt"
I would like to know the type of the flash you are using. such as which vendor.
(161 memory type : 64 capacity : 23)