Samsung/TizenRT #5911: master line  network driver can't be compiled through

Shark7Pepper, 2023-08-28:
Why is it that when I configure the network it doesn't compile, I chose to use the stm32. can you provide a config that will compile. Why are the functions arp_ipin(),ipv4_input() etc. used but not defined, this problem is causing the whole thing to fail to compile, how can I deal with it?

sunghan-chang, 2023-08-28:
@Shark7Pepper Could you share the configuration you used? The board is stm32??, configuration is ??

Shark7Pepper, 2023-08-28:
I configured the boards as stm32h745zi-nucleo/hello,Just added the network configuration.
#
# Automatically generated file; DO NOT EDIT.
# TinyAra Configuration
#

#
# Build Setup
#
# CONFIG_EXPERIMENTAL is not set
# CONFIG_DEFAULT_SMALL is not set
CONFIG_HOST_LINUX=y
# CONFIG_HOST_OSX is not set
# CONFIG_HOST_WINDOWS is not set
# CONFIG_HOST_OTHER is not set

#
# Build Configuration
#
CONFIG_APPS_DIR="../apps"
CONFIG_FRAMEWORK_DIR="../framework"
CONFIG_TOOLS_DIR="../tools"
CONFIG_BUILD_FLAT=y
# CONFIG_APP_BINARY_SEPARATION is not set
# CONFIG_BUILD_2PASS is not set
CONFIG_BOARD_BUILD_DATE="200204"

#
# Binary Output Formats
#
CONFIG_INTELHEX_BINARY=y
# CONFIG_MOTOROLA_SREC is not set
CONFIG_RAW_BINARY=y
# CONFIG_UBOOT_UIMAGE is not set
# CONFIG_DOWNLOAD_IMAGE is not set

#
# Customize Header Files
#
# CONFIG_ARCH_STDINT_H is not set
# CONFIG_ARCH_STDBOOL_H is not set
# CONFIG_ARCH_MATH_H is not set
# CONFIG_ARCH_FLOAT_H is not set
# CONFIG_ARCH_STDARG_H is not set
CONFIG_ARCH_HAVE_CUSTOMOPT=y
# CONFIG_DEBUG_NOOPT is not set
# CONFIG_DEBUG_CUSTOMOPT is not set
CONFIG_DEBUG_FULLOPT=y

#
# Chip Selection
#
CONFIG_ARCH_ARM=y
# CONFIG_ARCH_XTENSA is not set
CONFIG_ARCH="arm"
# CONFIG_ARCH_CHIP_LM is not set
# CONFIG_ARCH_CHIP_S5J is not set
# CONFIG_ARCH_CHIP_BCM4390X is not set
# CONFIG_ARCH_CHIP_IMX6 is not set
# CONFIG_ARCH_CHIP_STM32 is not set
# CONFIG_ARCH_CHIP_IMXRT is not set
# CONFIG_ARCH_CHIP_STM32L4 is not set
# CONFIG_ARCH_CHIP_AMEBAD is not set
CONFIG_ARCH_CHIP_STM32H745=y
# CONFIG_ARCH_CHIP_AMEBALITE is not set
# CONFIG_ARCH_CHIP_AMEBASMART is not set
CONFIG_ARCH_CHIP="stm32h745"
# CONFIG_ARM_THUMB is not set
# CONFIG_ARM_HAVE_DPFPU32 is not set
# CONFIG_ARM_HAVE_NEON is not set
[... 1170 lines cut ...]
#
# CONFIG_MESSAGING_IPC is not set

#
# Preference Support
#

#
# Things Management
#

#
# IoTBus Framework
#
# CONFIG_IOTBUS is not set

#
# Security Framework
#
CONFIG_SECURITY_API=y
# CONFIG_DEBUG_SECURITY_FRAMEWORK_ERROR is not set
# CONFIG_SECURITY_AUTH is not set
# CONFIG_SECURITY_CRYPTO is not set
# CONFIG_SECURITY_KEYMGR is not set
CONFIG_SECURITY_SS=y

sunghan-chang, 2023-08-28:
```
#
# Automatically generated file; DO NOT EDIT.
# TinyAra Configuration
#

#
# Build Setup
#
# CONFIG_EXPERIMENTAL is not set
# CONFIG_DEFAULT_SMALL is not set
CONFIG_HOST_LINUX=y
# CONFIG_HOST_OSX is not set
# CONFIG_HOST_WINDOWS is not set
# CONFIG_HOST_OTHER is not set

#
# Build Configuration
#
CONFIG_APPS_DIR="../apps"
CONFIG_FRAMEWORK_DIR="../framework"
CONFIG_TOOLS_DIR="../tools"
CONFIG_BUILD_FLAT=y
# CONFIG_APP_BINARY_SEPARATION is not set
# CONFIG_BUILD_2PASS is not set
CONFIG_BOARD_BUILD_DATE="200204"

#
# Binary Output Formats
#
CONFIG_INTELHEX_BINARY=y
# CONFIG_MOTOROLA_SREC is not set
CONFIG_RAW_BINARY=y
# CONFIG_UBOOT_UIMAGE is not set
# CONFIG_DOWNLOAD_IMAGE is not set

#
# Customize Header Files
#
# CONFIG_ARCH_STDINT_H is not set
# CONFIG_ARCH_STDBOOL_H is not set
# CONFIG_ARCH_MATH_H is not set
# CONFIG_ARCH_FLOAT_H is not set
# CONFIG_ARCH_STDARG_H is not set
CONFIG_ARCH_HAVE_CUSTOMOPT=y
# CONFIG_DEBUG_NOOPT is not set
# CONFIG_DEBUG_CUSTOMOPT is not set
CONFIG_DEBUG_FULLOPT=y

#
# Chip Selection
#
CONFIG_ARCH_ARM=y
# CONFIG_ARCH_XTENSA is not set
CONFIG_ARCH="arm"
# CONFIG_ARCH_CHIP_LM is not set
# CONFIG_ARCH_CHIP_S5J is not set
# CONFIG_ARCH_CHIP_BCM4390X is not set
# CONFIG_ARCH_CHIP_IMX6 is not set
# CONFIG_ARCH_CHIP_STM32 is not set
# CONFIG_ARCH_CHIP_IMXRT is not set
# CONFIG_ARCH_CHIP_STM32L4 is not set
# CONFIG_ARCH_CHIP_AMEBAD is not set
CONFIG_ARCH_CHIP_STM32H745=y
# CONFIG_ARCH_CHIP_AMEBALITE is not set
# CONFIG_ARCH_CHIP_AMEBASMART is not set
CONFIG_ARCH_CHIP="stm32h745"
# CONFIG_ARM_THUMB is not set
# CONFIG_ARM_HAVE_DPFPU32 is not set
# CONFIG_ARM_HAVE_NEON is not set
[... 1171 lines cut ...]
# CONFIG_MESSAGING_IPC is not set

#
# Preference Support
#

#
# Things Management
#

#
# IoTBus Framework
#
# CONFIG_IOTBUS is not set

#
# Security Framework
#
CONFIG_SECURITY_API=y
# CONFIG_DEBUG_SECURITY_FRAMEWORK_ERROR is not set
# CONFIG_SECURITY_AUTH is not set
# CONFIG_SECURITY_CRYPTO is not set
# CONFIG_SECURITY_KEYMGR is not set
CONFIG_SECURITY_SS=y
```

sunghan-chang, 2023-08-28:
@j11591 @hs36-kim Could you check this?

Shark7Pepper, 2023-08-28:
It seems that the eth driver for this board is not implemented.